espressif/esp-modbus #2: FreeModbus should be an optional component

brainstorm, 2022-04-05:
On any arbitrary esp-idf example, if one disables all Modbus-related components (TCP, RTU, ASCII), the result is the error below. Ideally the FreeModbus (esp-modbus) component should be completely independent of the rest and not occupy space nor compile time when building ModBus-unrelated IDF applications:

```C
(esp) rvalls@m1 esp_hid_device % cat sdkconfig | grep FMB
# CONFIG_FMB_COMM_MODE_TCP_EN is not set
# CONFIG_FMB_COMM_MODE_RTU_EN is not set
# CONFIG_FMB_COMM_MODE_ASCII_EN is not set
CONFIG_FMB_MASTER_TIMEOUT_MS_RESPOND=150
CONFIG_FMB_MASTER_DELAY_MS_CONVERT=200
CONFIG_FMB_QUEUE_LENGTH=20
CONFIG_FMB_PORT_TASK_STACK_SIZE=4096
CONFIG_FMB_SERIAL_BUF_SIZE=256
CONFIG_FMB_PORT_TASK_PRIO=10
CONFIG_FMB_PORT_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_FMB_CONTROLLER_SLAVE_ID_SUPPORT is not set
CONFIG_FMB_CONTROLLER_NOTIFY_TIMEOUT=20
CONFIG_FMB_CONTROLLER_NOTIFY_QUEUE_SIZE=20
CONFIG_FMB_CONTROLLER_STACK_SIZE=4096
CONFIG_FMB_EVENT_QUEUE_TIMEOUT=20
# CONFIG_FMB_TIMER_PORT_ENABLED is not set
# CONFIG_FMB_TIMER_USE_ISR_DISPATCH_METHOD is not set

(esp) rvalls@m1 esp_hid_device % idf.py build
Executing action: all (aliases: build)
Running ninja in directory /Users/rvalls/esp/esp-idf/examples/bluetooth/esp_hid_device/build
Executing "ninja all"...
[0/1] Re-running CMake...
-- Component directory /Users/rvalls/esp/esp-idf/components/cbor does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/rvalls/esp/esp-idf/components/coap does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/rvalls/esp/esp-idf/components/expat does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/rvalls/esp/esp-idf/components/libsodium does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/rvalls/esp/esp-idf/components/nghttp does not contain a CMakeLists.txt file. No component will be added
-- Building ESP-IDF components for target esp32c3
-- Project sdkconfig file /Users/rvalls/esp/esp-idf/examples/bluetooth/esp_hid_device/sdkconfig
Loading defaults file /Users/rvalls/esp/esp-idf/examples/bluetooth/esp_hid_device/sdkconfig.defaults...
Loading defaults file /Users/rvalls/esp/esp-idf/examples/bluetooth/esp_hid_device/sdkconfig.defaults.esp32c3...
-- App "esp_hid_device" version: v5.0-dev-2349-g4350e6fef8-dirty
-- Adding linker script /Users/rvalls/esp/esp-idf/examples/bluetooth/esp_hid_device/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /Users/rvalls/esp/esp-idf/components/esp_system/ld/esp32c3/sections.ld.in
-- Adding linker script /Users/rvalls/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.ld
-- Adding linker script /Users/rvalls/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.api.ld
-- Adding linker script /Users/rvalls/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.libgcc.ld
-- Adding linker script /Users/rvalls/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib.ld
-- Adding linker script /Users/rvalls/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.version.ld
-- Adding linker script /Users/rvalls/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib-time.ld
-- Adding linker script /Users/rvalls/esp/esp-idf/components/soc/esp32c3/ld/esp32c3.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp32c3 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_wifi espcoredump esptool_py fatfs freemodbus freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mdns mqtt newlib nvs_flash openthread partition_table protobuf-c protocomm pthread riscv sdmmc soc spi_flash spiffs tcp_transport tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant
-- Component paths: /Users/rvalls/esp/esp-idf/components/app_trace /Users/rvalls/esp/esp-idf/components/app_update /Users/rvalls/esp/esp-idf/components/asio /Users/rvalls/esp/esp-idf/components/bootloader /Users/rvalls/esp/esp-idf/components/bootloader_support /Users/rvalls/esp/esp-idf/components/bt /Users/rvalls/esp/esp-idf/components/cmock /Users/rvalls/esp/esp-idf/components/console /Users/rvalls/esp/esp-idf/components/cxx /Users/rvalls/esp/esp-idf/components/driver /Users/rvalls/esp/esp-idf/components/efuse /Users/rvalls/esp/esp-idf/components/esp-tls /Users/rvalls/esp/esp-idf/components/esp32c3 /Users/rvalls/esp/esp-idf/components/esp_adc_cal /Users/rvalls/esp/esp-idf/components/esp_common /Users/rvalls/esp/esp-idf/components/esp_eth /Users/rvalls/esp/esp-idf/components/esp_event /Users/rvalls/esp/esp-idf/components/esp_gdbstub /Users/rvalls/esp/esp-idf/components/esp_hid /Users/rvalls/esp/esp-idf/components/esp_http_client /Users/rvalls/esp/esp-idf/components/esp_http_server /Users/rvalls/esp/esp-idf/components/esp_https_ota /Users/rvalls/esp/esp-idf/components/esp_https_server /Users/rvalls/esp/esp-idf/components/esp_hw_support /Users/rvalls/esp/esp-idf/components/esp_lcd /Users/rvalls/esp/esp-idf/components/esp_local_ctrl /Users/rvalls/esp/esp-idf/components/esp_netif /Users/rvalls/esp/esp-idf/components/esp_phy /Users/rvalls/esp/esp-idf/components/esp_pm /Users/rvalls/esp/esp-idf/components/esp_ringbuf /Users/rvalls/esp/esp-idf/components/esp_rom /Users/rvalls/esp/esp-idf/components/esp_serial_slave_link /Users/rvalls/esp/esp-idf/components/esp_system /Users/rvalls/esp/esp-idf/components/esp_timer /Users/rvalls/esp/esp-idf/components/esp_wifi /Users/rvalls/esp/esp-idf/components/espcoredump /Users/rvalls/esp/esp-idf/components/esptool_py /Users/rvalls/esp/esp-idf/components/fatfs /Users/rvalls/esp/esp-idf/components/freemodbus /Users/rvalls/esp/esp-idf/components/freertos /Users/rvalls/esp/esp-idf/components/hal /Users/rvalls/esp/esp-idf/components/heap /Users/rvalls/esp/esp-idf/components/http_parser /Users/rvalls/esp/esp-idf/components/idf_test /Users/rvalls/esp/esp-idf/components/ieee802154 /Users/rvalls/esp/esp-idf/components/json /Users/rvalls/esp/esp-idf/components/log /Users/rvalls/esp/esp-idf/components/lwip /Users/rvalls/esp/esp-idf/examples/bluetooth/esp_hid_device/main /Users/rvalls/esp/esp-idf/components/mbedtls /Users/rvalls/esp/esp-idf/components/mdns /Users/rvalls/esp/esp-idf/components/mqtt /Users/rvalls/esp/esp-idf/components/newlib /Users/rvalls/esp/esp-idf/components/nvs_flash /Users/rvalls/esp/esp-idf/components/openthread /Users/rvalls/esp/esp-idf/components/partition_table /Users/rvalls/esp/esp-idf/components/protobuf-c /Users/rvalls/esp/esp-idf/components/protocomm /Users/rvalls/esp/esp-idf/components/pthread /Users/rvalls/esp/esp-idf/components/riscv /Users/rvalls/esp/esp-idf/components/sdmmc /Users/rvalls/esp/esp-idf/components/soc /Users/rvalls/esp/esp-idf/components/spi_flash /Users/rvalls/esp/esp-idf/components/spiffs /Users/rvalls/esp/esp-idf/components/tcp_transport /Users/rvalls/esp/esp-idf/components/tinyusb /Users/rvalls/esp/esp-idf/components/ulp /Users/rvalls/esp/esp-idf/components/unity /Users/rvalls/esp/esp-idf/components/usb /Users/rvalls/esp/esp-idf/components/vfs /Users/rvalls/esp/esp-idf/components/wear_levelling /Users/rvalls/esp/esp-idf/components/wifi_provisioning /Users/rvalls/esp/esp-idf/components/wpa_supplicant
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/rvalls/esp/esp-idf/examples/bluetooth/esp_hid_device/build
[511/1061] Performing build step for 'bootloader'
[0/1] Re-running CMake...
-- Component directory /Users/rvalls/esp/esp-idf/components/cbor does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/rvalls/esp/esp-idf/components/coap does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/rvalls/esp/esp-idf/components/expat does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/rvalls/esp/esp-idf/components/libsodium does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/rvalls/esp/esp-idf/components/nghttp does not contain a CMakeLists.txt file. No component will be added
-- Building ESP-IDF components for target esp32c3
-- Project sdkconfig file /Users/rvalls/esp/esp-idf/examples/bluetooth/esp_hid_device/sdkconfig
[... cut ...]
 eMBEventType
/Users/rvalls/esp/esp-idf/components/freemodbus/modbus/include/mb_m.h:409:29: error: unknown type name 'eMBMasterErrorEventType'; did you mean 'eMBEventType'?
 void vMBMasterSetErrorType( eMBMasterErrorEventType errorType );
                             ^~~~~~~~~~~~~~~~~~~~~~~
                             eMBEventType
[937/1059] Building CXX object esp-idf/asio/CMakeFiles/__idf_asio.dir/asio/asio/src/asio.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

igrr, 2022-04-06:
Hi @brainstorm, modbus component is being moved out of esp-idf to https://github.com/espressif/esp-modbus in the upcoming v5.0 release.

If you don't use modbus in the application, you can trim the build to not contain freemodbus component.

For example, you can do this by adding `set(EXCLUDE_COMPONENTS freemodbus)` to you project CMakeLists.txt, before the line where you include `project.cmake`.

You can also use `set(COMPONENTS main)` to trim the build even further, and then explicitly set component requirements (`REQUIRES` or `PRIV_REQUIRES` arguments of `idf_component_register`) in every component of your application (including `main`).

brainstorm, 2022-04-06:
Thanks Ivan, TIL ;)

closed 2022-04-06 by brainstorm