espressif/esp-modbus #5: Modbus is missing in docker image Espressif/idf:latest

goebela47, 2022-05-02:
Hi, 
the Espressif/idf:latest image (773b66c28144) provided on dockerhub.com does no longer contain the component "freemodbus".

Is this by accident or intention?

Best regards
Andreas

alisitsyn, 2022-05-02:
Hi Andreas,

This is intended behavior. The `freemodbus` component is no longer part of esp-idf and its image but will be supported as a separate [component](https://components.espressif.com/component/espressif/esp-modbus) in the component manager registry.

The component can be used with esp-idf v4.1 - v5.0.

goebela47, 2022-05-02:
Thank you, that link solves my issue.

igrr, 2022-05-02:
Just for reference in case someone else finds this issue, the docs related to IDF 5.0 migration can be found here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/removed-components.html