espressif/esp-modbus #6: rede pin on external io expander

Mr-Techtron, 2022-06-05:
I have a rede pin of rs485 pin connected to an external IO expander. Is there any way to control that rede pin through any callback function? I

alisitsyn, 2022-06-05:
Hi Mr-Techtron,

I am not sure that I understood your question correctly. Is the pin RE/DE of RS485 adapter connected to the external IO expender? If yes, The UART driver in RS485 mode can support only physical pins configured for UART. This is also to recommended way to provide RS485 driver control due to possible delays.

Mr-Techtron, 2022-06-06:
Thank you for your explanation.

alisitsyn, 2022-06-06:
If you need to provide the RS485 but do not have the free pin for RS485 direction control, you can use [auto direction ctrl option](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html#circuit-c-auto-switching-transmitter-receiver) approach. This will work just fine for relatively short distances.

Mr-Techtron, 2022-06-07:
[image]

Our board pin connection for rs485 is something like above. For the time being, I have implemented a simple modbus slave.

closed 2022-09-08 by alisitsyn