espressif/esp-modbus #8: run both tcp and rtu slave at same time

ahosny333, 2022-06-22:
hi,
is there any suggestions to run both RTU and TCP slave modbus at same time at same board ?

alisitsyn, 2022-06-23:
Hi @ahosny333,

Unfortunately, it is not possible to use RTU slave  + TCP slave on the same board for now. It is possible to use one instance of a master and one instance of a slave at the same time.

jmichelx2, 2022-09-12:
Hi,

Is is possible to run RTU master on UART1 and a RTU slave on UART2 in the same application?

Is there any future possibility to run two RTU masters on two UARTS in the same application?

alisitsyn, 2023-07-17:
This issue is related to https://github.com/espressif/esp-modbus/issues/9.
This feature request is in progress and will be implemented in modbus stack v2.0.0_beta1.

alisitsyn, 2024-03-05:
Closed as duplicate.

closed 2024-03-05 by alisitsyn