espressif/esp-modbus #9: port0 and port1 work with modbus RTU two instance of a slave at the same time

dorry3, 2022-07-12:
port0 and port1 work with modbus RTU two instance of a slave at the same time？

alisitsyn, 2022-07-12:
Hello @dorry3,

The esp-modbus supports one instance of master and one slave of each type to work simultaneously. It is not possible to start two Modbus RTU slaves at the same time.

Silvesterrr, 2023-01-28:
> Hello @dorry3,
> 
> The esp-modbus supports one instance of master and one slave of each type to work simultaneously. It is not possible to start two Modbus RTU slaves at the same time.

"of each type?" So does it means that now it's possible to have for example 1 rtu master, 1 rtu slave and 1 tcp slave simultaneously?

alisitsyn, 2023-01-30:
No, it was not clear enough. Only one master and one slave instance can work simultaneously in the application. For example: Slave RTU + Master TCP, Slave RTU + Master RTU, Master TCP + Slave RTU.

alisitsyn, 2023-07-17:
the status of it is in progress and will be implemented in stack v2.0.0.

closed 2024-01-24 by alisitsyn