espressif/esp-modbus #10: vMBMasterRunResRelease(231): Resource release failure

ph2355, 2022-07-25:
Hi, I'm constantly getting a vMBMasterRunResRelease(231): Resource release failure error in the log.
In my application I use polling for about 30 modbus registers every 5 seconds through modbus RTU.
The longer the app runs the more resource release failures I get through time.

So I did some digging where the error originates and got to vMBMasterRunResRelease function inside portevent_m.c file. 
As I understand, xMBMasterRunResTake is called before any modbus operation, and vMBMasterRunResRelease is called after. 
In vMBMasterRunResRelease function xEventGroupSetBits is called to set event bit, and then immediately check if the bit is set after xEventGroupSetBits returns. 
If the bit is not set, the function reports a resource release failure.

The problem here is that xEventGroupSetBits may not always return set bits even if the function has successfully set the desired bits.
If another task is already waiting for these event bits, xEventGroupSetBits will set those bits, then check for any waiting tasks, unblock these tasks and clear the bits. xEventGroupSetBits will in this case return cleared bits, so the error check in vMBMasterRunResRelease will fail even though the bits were successfully set.

I think this line does not work as intended and is pollutting my logs.
MB_PORT_CHECK((uxBits == MB_EVENT_RESOURCE), ; , "Resource release failure.");

alisitsyn, 2022-07-25:
@ph2355 ,

This is known thing that has been  originally commented and fixed in preliminary WIP version of modbus_support repo. It does not cause  any critical issue but agree should not be an error.

ph2355, 2022-07-29:
Thanks for the acknowledgement. Can I expect this to be fixed in a future version?

alisitsyn, 2022-07-29:
@ph2355 ,

Yes, sure this will be fixed and I will let you know once it merged. Thank you for the contribution.

brainstorm, 2022-10-24:
I second @ph2355 it's also cluttering my logs as I'm trying to debug fix issue #3 ... can you merge it in @alisitsyn?

alisitsyn, 2022-10-27:
@brainstorm, @ph2355 ,

The fix is merged in [commit](https://github.com/espressif/esp-modbus/commit/9ab2c68e84525e9ecbdaeff8b06e5e0e8df8e33a)

alisitsyn, 2023-10-24:
Close the issue as fixed.

closed 2023-10-24 by alisitsyn