espressif/esp-modbus #12: TCP transactions do not include the unit identifier field

AguileraG, 2022-12-09:
Hello,

The Modbus TCP specification defines a Unit Identifier field within the MBAP header, which is analogous to the Slave Address field used in Modbus RTU. This Unit ID field is necessary to communicate with TCP slaves that offer different data models in different Unit IDs, such as RTU-TCP gateways.

Currently, all TCP transactions use **MB_TCP_PSEUDO_ADDRESS** as a hard-coded Unit ID, which makes it impossible for a TCP master to communicate with these kind of devices. This behaviour could be fixed by using the real Unit ID inside the functions _eMBTCPSend_, _eMBTCPReceive_, _eMBMasterTCPSend_ and _eMBMasterTCPReceive_ and removing the **MB_TCP_PSEUDO_ADDRESS** define.

alisitsyn, 2022-12-09:
Hello, AguileraG,

Thank you for the issue. In original Modbus TCP protocol has two modifications: TCP and (RTU over TCP).
The Unit Identifier is used for RTU over TCP. For TCP communication the Unit Identifier is usually zero or 0xFF:
```
MBAP:
Unit Identifier (1 byte): This field is used to identify a remote
server located on a non TCP/IP network (for serial bridging). In a
typical Modbus TCP/IP server application, the unit ID is set to 00 or
FF, ignored by the server, and simply echoed back in the response.
```
The change with the MB_TCP_PSEUDO_ADDRESS is included in the code in order to avoid check of Unit Id on Modbus FSM layer. It was considered to delegate this check to related Port layer for now which has the required information and also checks the Transaction Identifier (TID). Some updates related to RTU over TCP are included but your issue is still actual and will be addressed later.

alisitsyn, 2023-03-01:
Hello AguileraG,

The changes to support UID have been merged in [v1.0.9](https://github.com/espressif/esp-modbus/releases/tag/v1.0.9).

Please take a look to the [issue](https://github.com/espressif/esp-idf/issues/10803) as well.

alisitsyn, 2023-07-17:
The issue will be closed as fixed. Please feel free to reopen if it happens again.

closed 2023-07-17 by alisitsyn