espressif/esp-modbus #13: UART haven’t APB Clock in esp32-c2

DreamSilverFox, 2022-12-30:
``` C
    uart_config_t xUartConfig = {
        .baud_rate = ulBaudRate,
        .data_bits = ucData,
        .parity = ucParity,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
        .rx_flow_ctrl_thresh = 2,
        .source_clk = UART_SCLK_APB
    };
```
esp32-c2 haven't **UART_SCLK_APB**.

In ESP32:
> UART_SCLK_APB
> UART_SCLK_REF_TICK
> UART_SCLK_DEFAULT

In ESP32-C2:
> UART_SCLK_PLL_F40M
> UART_SCLK_RTC
> UART_SCLK_XTAL
> UART_SCLK_DEFAULT

Can use **UART_SCLK_DEFAULT** inside of it?

DreamSilverFox, 2022-12-30:
This problem also in esp-idf 4.X(But in 3.x "source_clk" not be set).

alisitsyn, 2023-01-04:
@DreamSilverFox ,

Thank you for your issue. Please note that the esp-modbus component is actually suppots esp-idf v4.1 - v5.1 and the esp32c2 support is not merged yet. However, the code can be simply modified right as you suggested above and you can use the the clock option of `UART_SCLK_DEFAULT`.
I suggest to use the macro to check the IDF version as below:

port/portserial_m.c:
```
    uart_config_t xUartConfig = {
        .baud_rate = ulBaudRate,
        .data_bits = ucData,
        .parity = ucParity,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
        .rx_flow_ctrl_thresh = 2,
#if (ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(5, 0, 0))
        .source_clk = UART_SCLK_DEFAULT,
#else
        .source_clk = UART_SCLK_APB
#endif
    };   
```
It is also required to change the file main/Kconfig.projbuild.
Some tricks are also possible to compile the modbus under v3.3, but this is not recommended as obsolete.
Please let me know some specific information about your project if you need more help.

Thank you.

alisitsyn, 2023-01-11:
Merged in commit aca48fa7

closed 2023-03-07 by alisitsyn