espressif/esp-modbus #14: Memory leak in Modbus master initialization/teardown

snmcbee, 2023-01-09:
Calling `mbc_master_init()` creates a binary semaphore for the receive buffer in `xMBMasterPortRxSemaInit()`, which makes a dynamic memory allocation. `mbc_master_destroy()` does not destroy this semaphore, resulting in a memory leak each time the Modbus master is re-initialized. This could be resolved by calling `vSemaphoreDelete()` in the destroy function, or by creating a statically allocated semaphore.

alisitsyn, 2023-01-09:
@nicksenva ,

Thank you for this issue. The originally it supposed to be static semaphore. The code is updated as in the patch below and will be merged ASAP.
[0001-fix-rx-sema-destroy.patch](https://github.com/espressif/esp-modbus/files/10375933/0001-fix-rx-sema-destroy.patch.log)

nicksenva, 2023-01-09:
Thank you for your quick response. For now we have applied a patch by moving our esp-modbus component out of our project's managed components folder. Is there any estimate when the next official release will include this patch?

alisitsyn, 2023-01-10:
I can not give you the concrete date of merge. However it will be merged as soon as other pending MRs are tested. I will try to do it ASAP.

alisitsyn, 2023-01-12:
The fix has been merged in commit 3f6537fc940c932884db43c9d62608e982461a70

closed 2023-03-07 by alisitsyn