espressif/esp-modbus #17: Modbus Master Crashes

fink-at-trmc-dk, 2023-01-14:
Hi again,

I get the following in the terminal after running the below code for som time. I poll a slave with UID 71 (Kostal Inverter) periodically every  10 seconds.
```
I (887486) MB_CONTROLLER_MASTER: modbus_tcp_mast 2112
I (887496) ModbusTCP: state_t::RELEASE:
I (887506) ModbusTCP: ModbusTCP 5972
I (897506) MB_CONTROLLER_MASTER: modbus_tcp_mast 2112
I (897506) MB_CONTROLLER_MASTER: modbus_tcp_mast 2112
E (898506) MB_CONTROLLER_MASTER: mbc_master_get_parameter(73): Master get parameter failure, error=(0x108) (ESP_ERR_INVALID_RESPONSE).
I (898506) MB_CONTROLLER_MASTER: modbus_tcp_mast 2112
I (898516) MB_CONTROLLER_MASTER: modbus_tcp_mast 2112
I (898516) MB_CONTROLLER_MASTER: modbus_tcp_mast 2112
I (899516) MB_CONTROLLER_MASTER: modbus_tcp_mast 2112
E (899516) MB_CONTROLLER_MASTER: mbc_master_get_parameter(73): Master get parameter failure, error=(0x108) (ESP_ERR_INVALID_RESPONSE).
I (899516) ModbusTCP: state_t::RELEASE:
I (899526) ModbusTCP: ModbusTCP 5972
I (905516) MB_CONTROLLER_MASTER: modbus_tcp_mast 2112
E (909526) MB_CONTROLLER_MASTER: mbc_master_get_parameter(73): Master get parameter failure, error=(0x108) (ESP_ERR_INVALID_RESPONSE).
I (909526) MB_CONTROLLER_MASTER: modbus_tcp_mast 2112
I (909526) MB_CONTROLLER_MASTER: modbus_tcp_mast 2112
I (910526) MB_CONTROLLER_MASTER: modbus_tcp_mast 2112
E (910526) MB_CONTROLLER_MASTER: mbc_master_get_parameter(73): Master get parameter failure, error=(0x108) (ESP_ERR_INVALID_RESPONSE).
I (910526) MB_CONTROLLER_MASTER: modbus_tcp_mast 2112
I (910536) ModbusTCP: state_t::RELEASE:
I (910536) MB_CONTROLLER_MASTER: modbus_tcp_mast 2112
I (910546) ModbusTCP: ModbusTCP 5972
I (910726) MB_CONTROLLER_MASTER: modbus_tcp_mast 2112
I (910726) MB_CONTROLLER_MASTER: modbus_tcp_mast 2112
I (910726) MB_CONTROLLER_MASTER: modbus_tcp_mast 2112
I (920556) ModbusTCP: CID_SET_CHARGE (0.000000) read successful.
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008e0e2  PS      : 0x00060233  A0      : 0x8008daa5  A1      : 0x3ffd0520
0x4008e0e2: remove_free_block at C:/espressif/esp-idf/components/heap/tlsf/tlsf.c:332
 (inlined by) block_locate_free at C:/espressif/esp-idf/components/heap/tlsf/tlsf.c:567
 (inlined by) tlsf_malloc at C:/espressif/esp-idf/components/heap/tlsf/tlsf.c:1004

A2      : 0x3ffb97f4  A3      : 0x0000000c  A4      : 0x3ffb984c  A5      : 0x00000000
A6      : 0x00000003  A7      : 0x0000cdcd  A8      : 0x3ffb0005  A9      : 0x00000003
A10     : 0x3ffb97f4  A11     : 0x00000000  A12     : 0x00b33fff  A13     : 0x00060223
A14     : 0x400e9388  A15     : 0x0000cdcd  SAR     : 0x0000001d  EXCCAUSE: 0x0000001d
0x400e9388: lwip_select at C:/espressif/esp-idf/components/lwip/lwip/src/api/sockets.c:1972

EXCVADDR: 0x00b3400b  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb

Backtrace: 0x4008e0df:0x3ffd0520 0x4008daa2:0x3ffd0540 0x40082091:0x3ffd0560 0x400820f1:0x3ffd0580 0x40082126:0x3ffd05a0 0x4008ffe0:0x3ffd05c0 0x40090004:0x3ffd05e0 0x400d77de:0x3ffd0600 0x400debf9:0x3ffd0660 0x400dfbb8:0x3ffd06b0 0x400dfc6c:0x3ffd06e0 0x400e02ff:0x3ffd0710 0x400dfef1:0x3ffd0730 0x400de782:0x3ffd0770 0x4008b9c9:0x3ffd07d0
0x4008e0df: remove_free_block at C:/espressif/esp-idf/components/heap/tlsf/tlsf.c:331
 (inlined by) block_locate_free at C:/espressif/esp-idf/components/heap/tlsf/tlsf.c:567
 (inlined by) tlsf_malloc at C:/espressif/esp-idf/components/heap/tlsf/tlsf.c:1004

0x4008daa2: multi_heap_malloc_impl at C:/espressif/esp-idf/components/heap/multi_heap.c:207

0x40082091: heap_caps_malloc_base at C:/espressif/esp-idf/components/heap/heap_caps.c:145

0x400820f1: heap_caps_malloc at C:/espressif/esp-idf/components/heap/heap_caps.c:165

0x40082126: heap_caps_malloc_default at C:/espressif/esp-idf/components/heap/heap_caps.c:190

0x4008ffe0: _calloc_r at C:/espressif/esp-idf/components/newlib/heap.c:65

0x40090004: calloc at C:/espressif/esp-idf/components/newlib/heap.c:29

0x400d77de: esp_vfs_select at C:/espressif/esp-idf/components/vfs/vfs.c:980

0x400debf9: xMBTCPPortMasterCheckAlive at C:/projects/sntp/components/espressif__esp-modbus/freemodbus/tcp_master/port/port_tcp_master.c:419 (discriminator 3)

0x400dfbb8: xMBMasterTCPPortWritePoll at C:/projects/sntp/components/espressif__esp-modbus/freemodbus/tcp_master/port/port_tcp_master.c:958

[... 209 lines cut ...]
	case state_t::RELEASE:
	{
		ESP_LOGI(TAG, "state_t::RELEASE:");

		if (now > _times.timeCharge)
		{
			_state = state_t::CHARGE;
		}
		break;
	}
	}
}

void ModbusTCP::SetTimes(times_t times)
{
	xQueueSend(modbusQueue, &times, 10000 / portTICK_PERIOD_MS);
}
```

Does anyone have an idea as to how to fix it or go about debugging it? I'm unsure as to where to start...

Please net me know if more information is needed :-)

BR
Fink

alisitsyn, 2023-01-14:
Hi @fink-at-trmc-dk ,

The formal process is started for the issue. I will let you know the result as soon as possible. Your `sdkconfig, map, elf, bin, srcs (if possible)` files can be useful. Thanks.

fink-at-trmc-dk, 2023-01-14:
Hi again and thanks for the fast response. I have attached a zip with the project which is a WIP, but the failure remains.
I have a diagslave.exe running as ```diagslave.exe -c 100 -a71 -o1 -p1502``` which means that I have modified ```xMBMasterTCPPortSendResponse line 990``` by adding ```pucMBTCPFrame[MB_TCP_UID] = (UCHAR)(pxInfo->ucSlaveAddr);``` to have the slave ID correctly set.

As seen in the log it may take some time to crash. Last log entry on my machine was nearly 900 seconds.

Please let me know if you need more from me :-)

//Fink
[sntp.zip](https://github.com/espressif/esp-modbus/files/10418985/sntp.zip)

alisitsyn, 2023-01-17:
I tried  to reproduce the issue using your whole project. My modifications are: 

1. SSID, Password of wifi
2. IP Address of the Modbus master slave.
3. changes in `xMBMasterTCPPortSendResponse line 990` that are required for diagslave.
4. compile project on linux machine and use diagslave on linux.

The project has been checked with the modbus component from your components folder then with esp-modbus v1.0.7, then with v1.0.8. 
The modbus slave: /home/esp-man/diagslave/x86_64-linux-gnu/diagslave -c 100 -a71 -o2 -p1502

The extended logs are attached:

[sntp.log](https://github.com/espressif/esp-modbus/files/10435642/sntp.log)
[log_diagslave.txt](https://github.com/espressif/esp-modbus/files/10435643/log_diagslave.txt)

The testing was performed for couple of hours without any crash. Please let me know which version of esp-idf you are using and try the same with master of esp-idf.

Thanks.

fink-at-trmc-dk, 2023-01-18:
Hi again,
Thanks for taking your time on this issue.
I updated ESP-IDF to lates version from Main instead of V5.0. Regarding
Modbus I could not find a tag v1.0.8 and do not know what commit is the
best v1.0.8, so I also took the latest from master here and added changes
in `xMBMasterTCPPortSendResponse` to enable specific slave ID.
It has been running for several hours (> 10) without any hickups so the
problem seems to have solved itself with the bleeding edge of both IDF and
Modbus.

Is it possible for you in the future to use the tags so it is easy to
checkout and use a specific version like in the rest of the ESP-IDF
environment?

In any case: Thanks for the help :-)
Case can be closed from my perspective.
//Fink

alisitsyn, 2023-01-18:
Hi, 

Thank you for the update. It is always good when solution is found.
The latest version of esp-modbus is v[1.0.8](https://components.espressif.com/components/espressif/esp-modbus) and there are no any changes that fixes any issue related to what you described. You can select this version in your manifest file or just download the archive, extract it to the components folder and try to test. The changes related to UID will be merged with some other pending MRs. I will try to do this as soon as possible.

> Is it possible for you in the future to use the tags so it is easy to
checkout and use a specific version like in the rest of the ESP-IDF
environment?

In the internal system the version tag is applied before each merge. Once the version is merged the version is being synchronized with the giithub repo and uploaded to component manager. I will check the project rules to apply the version tag on the github project accordingly.
 
I agree the issue can be closed then. Feel free to reopen the issue if you have some additional results. 
Thanks.

closed 2023-03-07 by alisitsyn