espressif/esp-modbus #20: Modbus Master RTU parity errors not detected

nicksenva, 2023-02-01:
Our device acting as a Modbus master does not appear to check parity values on receive frames from the slave device. 
In our test setup, the ESP master is configured for even parity and connected to a slave test device which transmits back with odd parity. However, the master does not recognize the parity error and calls to mbc_master_send_request() return without error. This also occurs when the master and slave are configured for even and odd parity, respectively.

Here is a capture of the master sending a valid request frame with even parity:
[image]

And here is the slave's invalid response which is still accepted:
[image]

The only condition in which errors are returned is when an unexpected number of bits is received (i.e. master is configured for no parity bits but slave returns even/odd parity, or the slave returns an additional stop bit).

alisitsyn, 2023-02-02:
Hello,

Please send here your sdkconfig file and your code to setup master in order to reproduce the issue.
Thank you.

nicksenva, 2023-02-02:
Hello,

Here is the isolated Modbus test code from our project.
```
#include "string.h"
#include "esp_log.h"
#include "mbcontroller.h"
#include "sdkconfig.h"
#include "esp_modbus_common.h"

static const char *TAG = "MASTER_TEST";

#define MB_UART_PORT_NUM 1
#define MB_UART_TXD 32
#define MB_UART_RXD 35
#define MB_UART_RTS 33
#define MB_BAUD 115200
#define MB_STOP_BITS UART_STOP_BITS_1

esp_err_t modbus_get(uint16_t start_reg, uint16_t nRegs, uint16_t reg_type, void* dstBuf) {
	esp_err_t err;
	double slave_addr = 31;
	err = mbc_master_send_request(
		&(mb_param_request_t) {slave_addr, reg_type, start_reg, nRegs}, dstBuf);

	return err;
}

esp_err_t modbus_init() {
	// Initialize and start Modbus controller
	mb_communication_info_t comm = {
			.port = MB_UART_PORT_NUM,
			.mode = MB_MODE_RTU,
			.baudrate = MB_BAUD,
			.parity = UART_PARITY_EVEN
	};
	void* master_handler = NULL;
	esp_err_t err;

	err = mbc_master_init(MB_PORT_SERIAL_MASTER, &master_handler);
	MB_RETURN_ON_FALSE((master_handler != NULL), ESP_ERR_INVALID_STATE, TAG,
								"mb controller initialization fail.");
	MB_RETURN_ON_FALSE((err == ESP_OK), ESP_ERR_INVALID_STATE, TAG,
							"mb controller initialization fail, returns(0x%"PRIu32").",
							(uint32_t)err);
	err = mbc_master_setup((void*)&comm);
	MB_RETURN_ON_FALSE((err == ESP_OK), ESP_ERR_INVALID_STATE, TAG,
							"mb controller setup fail, returns(0x%"PRIu32").",
							(uint32_t)err);
	err = uart_set_pin(MB_UART_PORT_NUM, MB_UART_TXD, MB_UART_RXD,
			 	 	 	 MB_UART_RTS, UART_PIN_NO_CHANGE);
	MB_RETURN_ON_FALSE((err == ESP_OK), ESP_ERR_INVALID_STATE, TAG,
			"mb serial set pin failure, uart_set_pin() returned (0x%"PRIu32").", (uint32_t)err);

	err = mbc_master_start();
	MB_RETURN_ON_FALSE((err == ESP_OK), ESP_ERR_INVALID_STATE, TAG,
							"mb controller start fail, returns(0x%"PRIu32").",
							(uint32_t)err);

	// Set driver mode to Half Duplex
	err = uart_set_mode(MB_UART_PORT_NUM, UART_MODE_RS485_HALF_DUPLEX);
	MB_RETURN_ON_FALSE((err == ESP_OK), ESP_ERR_INVALID_STATE, TAG,
			"mb serial set mode failure, uart_set_mode() returned (0x%"PRIu32").", (uint32_t)err);

	ESP_LOGI(TAG, "Modbus master stack initialized...");

	// Set UART stop bits
   err = uart_set_stop_bits(MB_UART_PORT_NUM, MB_STOP_BITS);
   MB_RETURN_ON_FALSE((err == ESP_OK), ESP_ERR_INVALID_STATE, TAG,
		   "mb serial set stop bits failure, uart_set_stop_bits() returned (0x%"PRIu32").", (uint32_t)err);
[... 195 lines cut ...]
D (964) MB_PORT_COMMON: xMBPortSerialWaitEvent, UART event: 5
D (974) MB_MASTER_SERIAL: MB_uart[1] event:
D (974) MB_MASTER_SERIAL: uart parity error.
D (984) MB_PORT_COMMON: xMBPortSerialWaitEvent, UART event: 0
D (984) MB_MASTER_SERIAL: MB_uart[1] event:
D (994) MB_MASTER_SERIAL: Data event, len: 7.
D (994) MB_MASTER_SERIAL: Received data: 8(bytes in buffer)

D (1004) MB_MASTER_SERIAL: Timeout occured, processed: 8 bytes
D (1004) MB_PORT_COMMON: eMBMasterPoll:EV_MASTER_FRAME_SENT
D (1014) POLL sent buffer: 03 00 01 00 01
D (1014) MB_PORT_COMMON:  xMBMasterPortSerialGetResponse default
D (1024) MB_PORT_COMMON: eMBMasterRTUReceive: Port enter critical.
D (1024) MB_PORT_COMMON: eMBMasterRTUReceive: Port exit critical
D (1034) MB_PORT_COMMON: eMBMasterPoll: Packet data received successfully (0).
D (1044) POLL receive buffer: 03 02 02 d4
D (1044) MB_PORT_COMMON: eMBMasterPoll:EV_MASTER_EXECUTE
D (1054) MB_PORT_COMMON: eMBMasterPoll: set event EV_ERROR_OK
D (1054) MB_PORT_COMMON: eMBMasterPoll:EV_MASTER_ERROR_PROCESS
D (1064) MB_PORT_COMMON: vMBMasterCBRequestSuccess: Callback request success.
D (1074) MB_PORT_COMMON: eMBMasterWaitRequestFinish: returned event = 0x40
I (1074) MASTER_TEST: Val read: 724
```

[sdkconfig.txt](https://github.com/espressif/esp-modbus/files/10572463/sdkconfig.txt)

nicksenva, 2023-02-08:
After reviewing the source code we have discovered that the low-level UART driver is detecting the parity error in its receive ISR handler, which sends an event notification to the UART event task. This task has a case statement for logging the error, but it doesn't do anything to handle it. Is there some other location in the code where parity errors are handled, and if so what is the defined behavior? I would expect the error to be propagated and returned to the function sending the request. Is this correct?
[image]

alisitsyn, 2023-02-09:
@nicksenva ,

Thank you for your update and sorry for the delay with answer. The Modbus code should not handle the parity error in the event task. The event task just catch it to log the error message. The handling is performed in the [RTU handling code](https://github.com/espressif/esp-modbus/blob/master/freemodbus/modbus/rtu/mbrtu_m.c#L177). If the parity error happens the received frame will not contain the correct CRC this causes the FSM to trigger error handling and the error `ESP_ERR_INVALID_RESPONSE` at the end. I will try to reproduce your error as soon as I can.

ohitsderrick, 2023-02-09:
@alisitsyn, I think the problem is that it will not fail if the parity is intentionally configured incorrectly (transmitter and receiver have different parity settings). During normal operation when there is no actual corruption, the raw data is correct and passes the CRC, but the parity errors are ignored and the UART driver still passes data to the Modbus layer as if if everything is fine.

alisitsyn, 2023-02-09:
@ohitsderrick,

Yes, this sounds correct in this case. The frame check is applied regardless of any parity checking method used for the individual
characters of the message. The parity check of UART has to apply to the error check result. 

in order to fix this error it is enough just to reset the buffer in case of parity error.
``` c
                //Event of UART parity check error
                case UART_PARITY_ERR:
                    ESP_LOGD(TAG, "uart parity error.");
                    xQueueReset(xMbUartQueue);
                    uart_flush_input(ucUartNumber);
                    break;
```

nicksenva, 2023-02-09:
The patch appears to work as intended after a few minutes of testing, here is a capture showing the behavior with a parity mismatch. For now we will continue to use the buffer reset as a quick fix until an official patch is made. 

```
D (703610) MB_PORT_COMMON: xMBMasterRunResTake:Take resource (80) (101 ticks).
D (703620) MB_PORT_COMMON: eMBMasterPoll:EV_MASTER_FRAME_TRANSMIT
D (703630) MB_PORT_COMMON: xMBMasterPortSerialSendRequest default
D (703630) MB_PORT_COMMON: vMBMasterPortTimersRespondTimeoutEnable Respond enable timeout.
D (703640) MB_MASTER_SERIAL: MB_TX_buffer sent: (9) bytes.
D (703650) MB_PORT_COMMON: xMBPortSerialWaitEvent, UART event: 1
D (703650) MB_MASTER_SERIAL: MB_uart[1] event:
D (703660) MB_MASTER_SERIAL: uart rx break.
D (703660) MB_PORT_COMMON: xMBPortSerialWaitEvent, UART event: 1
D (703670) MB_MASTER_SERIAL: MB_uart[1] event:
D (703670) MB_MASTER_SERIAL: uart rx break.
D (703680) MB_PORT_COMMON: xMBPortSerialWaitEvent, UART event: 5
D (703680) MB_MASTER_SERIAL: MB_uart[1] event:
D (703690) MB_MASTER_SERIAL: uart parity error, count: 14.
D (703690) MB_PORT_COMMON: xMBPortSerialWaitEvent, UART event: 5
D (703700) MB_MASTER_SERIAL: MB_uart[1] event:
D (703700) MB_MASTER_SERIAL: uart parity error, count: 15.
D (703710) MB_MASTER_SERIAL: xMBMasterPortRxSemaTake: RX semaphore take failure.
D (703720) MB_PORT_COMMON: xMBPortSerialWaitEvent, UART event: 5
D (703720) MB_MASTER_SERIAL: MB_uart[1] event:
D (703730) MB_MASTER_SERIAL: uart parity error, count: 16.
D (703730) MB_MASTER_SERIAL: xMBMasterPortRxSemaTake: RX semaphore take failure.
D (703740) MB_PORT_COMMON: xMBPortSerialWaitEvent, UART event: 5
D (703750) MB_MASTER_SERIAL: MB_uart[1] event:
D (703750) MB_MASTER_SERIAL: uart parity error, count: 17.
D (703760) MB_MASTER_SERIAL: xMBMasterPortRxSemaTake: RX semaphore take failure.
D (703760) MB_PORT_COMMON: xMBPortSerialWaitEvent, UART event: 5
D (703770) MB_MASTER_SERIAL: MB_uart[1] event:
D (703770) MB_MASTER_SERIAL: uart parity error, count: 18.
D (703780) MB_MASTER_SERIAL: xMBMasterPortRxSemaTake: RX semaphore take failure.
D (703790) MB_PORT_COMMON: xMBPortSerialWaitEvent, UART event: 5
D (703790) MB_MASTER_SERIAL: MB_uart[1] event:
D (703800) MB_MASTER_SERIAL: uart parity error, count: 19.
D (703800) MB_MASTER_SERIAL: xMBMasterPortRxSemaTake: RX semaphore take failure.
D (703810) MB_PORT_COMMON: xMBPortSerialWaitEvent, UART event: 5
D (703820) MB_MASTER_SERIAL: MB_uart[1] event:
D (703820) MB_MASTER_SERIAL: uart parity error, count: 20.
D (703830) MB_MASTER_SERIAL: xMBMasterPortRxSemaTake: RX semaphore take failure.
D (703830) MB_PORT_COMMON: xMBPortSerialWaitEvent, UART event: 0
D (703840) MB_MASTER_SERIAL: MB_uart[1] event:
D (703850) MB_MASTER_SERIAL: Data event, len: 7.
D (703850) MB_PORT_COMMON: eMBMasterPoll:EV_MASTER_FRAME_SENT
D (703860) MB_PORT_COMMON: eMBMasterPoll:EV_MASTER_ERROR_PROCESS
D (703860) MB_PORT_COMMON: vMBMasterErrorCBReceiveData:Callback receive data timeout failure.
D (703870) MB_PORT_COMMON: eMBMasterWaitRequestFinish: returned event = 0x100
E (703880) MB_CONTROLLER_MASTER: mbc_master_send_request(93): Master send request failure error=(0x108) (ESP_ERR_INVALID_RESPONSE).
```

On a side note, our esp-modbus component is in our project's components folder rather than the managed components so we can make modifications as necessary. We would like to clone the GitHub repo so we can pull the latest releases, but the release tags in the component registry aren't present on the Git repo. Are there any plans to keep these tags in sync?

alisitsyn, 2023-02-13:
@ nicksenva ,

Thank you for the test results.

> Here is the log output for this condition. We have noticed that every read request reports there being two UART break events even when no response is returned from the slave. We also noticed that the send/received byte count are both offset by one (9 bytes reported sent in an 8 byte frame, and 8 bytes received in a 7 byte response. Could you explain these discrepancies?

The break events are related to hardware issue in UART working in RS485 mode. On the break event the RX FIFO can receive one zero byte in this case which is filtered out. The debug messages shows the offset in the buffer instead of the length of the message and are misleading. It is related to workaround for UART issues.

> Specifically, when the master UART is configured for 2 stop bits and the slave is configured for 1 stop bit, the framing error is not detected (or should this be a break error?).

I think the break events are related to situation described above but not related to incorrect framing detection. 
For example: I the master is configured for 1 stop it won't have any issue for receiver - should the slave be configured for 2 stop bits you will already completely receive each character after 1 stop bit and the second will simply be an additional 1 bit period pause (IDLE). The issues can happen If the master is configured to 1 stop and slave to 2 stop bits when you send two characters "immediately" after another and the start bit of the second character is sent after just 1 stop bit time, while the slave is still waiting for the second stop bit to arrive. In the worst case the slave's receiver will see your start bit as an invalid second stop bit and then also miss the start bit and hence the next character will be corrupted. However the peer, while working in 2 stop bit mode may also simply accept that there was a fast new start bit and so not have any problems. Therefore what actually happens is undefined. The UART driver does use the interrupt instead of DMA and does not detect any frame error in this case as per my test results. This can be investigated more but I would not say it is a critical discrepancy. If you disagree with this I would propose you to create new issue against UART driver in esp-idf. 
The issue with break conditions are reported to digital team but will need time for investigation. Also, it is better to create the separate esp-idf issue for this behavior as well to be able to track it appropriately.

> would like to clone the GitHub repo so we can pull the latest releases, but the release tags in the component registry aren't present on the Git repo. Are there any plans to keep these tags in sync?
 
The esp-modbus project rules include tagging of major updates only, this is why the latest updates do not contain the tags. Not an issue, I can apply the tags and sync them with github repo.

alisitsyn, 2023-03-07:
The simplified fix has been implemented in commit 8e3771847f826aad37491b96b2b4a2de8d2cbf76.
The issue will be closed. Feel free to reopen the issue. The separated issue will be tracked as per esp-idf IDFGH-9386.

closed 2023-03-16 by alisitsyn