espressif/esp-modbus #22: After switching to ESP-IDF 5.0.0, all projects using Modbus RTU stopped compiling

kotyara12, 2023-02-15:
After switching to ESP-IDF 5.0.0, all projects using Modbus RTU stopped compiling
It is completely incomprehensible to me why it was necessary to break the working environment
Downloaded this repo but it flatly refuses to compile

```
.pio/libdeps/esp32dev/esp-modbus/freemodbus/common/esp_modbus_master_tcp.c:8:10: fatal error: esp_modbus_master.h: No such file or directory

***************************************************************************
* Looking for esp_modbus_master.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_modbus_master.h"
* Web  > https://registry.platformio.org/search?q=header:esp_modbus_master.h
*
***************************************************************************

    8 | #include "esp_modbus_master.h"      // for public interface defines
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from .pio/libdeps/esp32dev/esp-modbus/freemodbus/common/esp_modbus_master.c:8:
.pio/libdeps/esp32dev/esp-modbus/freemodbus/common/mbc_master.h:18:10: fatal error: esp_modbus_common.h: No such file or directory

***************************************************************************
* Looking for esp_modbus_common.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_modbus_common.h"
* Web  > https://registry.platformio.org/search?q=header:esp_modbus_common.h
*
***************************************************************************

   18 | #include "esp_modbus_common.h"      // for common types
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from .pio/libdeps/esp32dev/esp-modbus/freemodbus/common/esp_modbus_master_serial.c:8:
.pio/libdeps/esp32dev/esp-modbus/freemodbus/common/mbc_master.h:18:10: fatal error: esp_modbus_common.h: No such file or directory

***************************************************************************
* Looking for esp_modbus_common.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_modbus_common.h"
* Web  > https://registry.platformio.org/search?q=header:esp_modbus_common.h
*
***************************************************************************

   18 | #include "esp_modbus_common.h"      // for common types
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32dev\libadd\esp-modbus\freemodbus\common\esp_modbus_master_tcp.o] Error 1
*** [.pio\build\esp32dev\libadd\esp-modbus\freemodbus\common\esp_modbus_master.o] Error 1
*** [.pio\build\esp32dev\libadd\esp-modbus\freemodbus\common\esp_modbus_master_serial.o] Error 1
============================================================================ [FAILED] Took 397.66 seconds ============================================================================

```

alisitsyn, 2023-02-15:
Hello @kotyara12,
This compilation issue related to the freemodbus component moved to separate [repository](https://github.com/espressif/esp-modbus) from esp-idf v5.0.0. Please see this change mentioned in [migration guide](https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32/migration-guides/release-5.x/protocols.html#id9) and how to migrate to new esp-modbus component.
In order to repair your project you need just to add the manifest file to your `pio_project_root_dir/src/` folder as below:
```
dependencies:
  idf: ">=4.1"
  espressif/esp-modbus:
    version: "^1.0" 
```
Then select in platformio: "PROJECT TASKS" - > "esp32dev" -> "General" -> "Clean All" -> "Build"
The issue should be solved. If you still have the issues please place the compilation log here.
[idf_component.yml.log (remove log ext before use)](https://github.com/espressif/esp-modbus/files/10741079/idf_component.yml.log)

Thanks.

kotyara12, 2023-02-15:
Thank you very much, this solved the problem.
Sorry for being too emotional, but there are already a lot of surprises in this migration. There are a lot of things to rewrite.

closed 2023-02-15 by kotyara12

alisitsyn, 2023-02-15:
> Thank you very much, this solved the problem. Sorry for being too emotional, but there are already a lot of surprises in this migration. There are a lot of things to rewrite.

I understand this. No problem, feel free to submit an issue if you have any problems. Good luck!