espressif/esp-modbus #24: How to Read Device Identification Function 43 / 14 (0x2B / 0x0E) with MODBUS Master RTU

joaoviktor22, 2023-03-15:
Hello,

I'm trying to get the ID of my slave using my MODBUS Master RTU, but I'm having trouble reading the Device Identification with Function 43 and 03 (Extended Objects). I've searched extensively, but haven't found a solution yet. If anyone knows how to do this, I would greatly appreciate your help.

alisitsyn, 2023-03-20:
Hello @joaoviktor22,

The Modbus `Function 43: Read Device Identification` is not supported by the stack. 
Usually vendors using the function 17 (0x11) Report Slave ID command returning a free-form string of vendor/model/serial-number, instead of the Read Device Identification which s supported. 
As per [Modbus specification]( "https://modbus.org/docs/Modbus_Application_Protocol_V1_1b.pdf) page 43, the stack requires to implement an encapsulated Modbus interface with composed data elements (objects). 
The example of implementation can be taken from [libmodbus library](https://github.com/stephane/libmodbus/tree/read-device-id).

joaoviktor22, 2023-04-05:
Thank you for your response. I had originally intended to use function 43, but I was able to resolve the issue by directly implementing UART communication. Thank you for your assistance.

alisitsyn, 2023-04-06:
@joaoviktor22 ,

Thank you for update and this feature request. This command will be added to the feature list with low priority.

alisitsyn, 2025-04-03:
@joaoviktor22 ,

The function <0x2B Read Device Identification> will not be implemented as default function. However, the [Custom command  feature in v2 has been merged](https://github.com/espressif/esp-modbus/pull/101). You can use the feature to implement this function with the handler.

closed 2025-04-03 by alisitsyn