espressif/esp-modbus #25: How to add word_length & stop_bits in mb_communication_info_t struct

JimitD, 2023-03-24:
Hello Team, 

In '`mb_communication_info_t stuct`' there are no member given to configure '`word_length`' and '`stop_bits`'. 

Where should I add this parameters to function accordingly. 

` comm_info.mode = MB_MODE_RTU,
    comm_info.slave_addr = SALVE_ADDR_;
    comm_info.port = MB_PORT_NUM;
    comm_info.baudrate = BAUDRATE_;
    comm_info.parity = PARITY_;
    esp_err_t err_slave_setup = ESP_ERROR_CHECK_WITHOUT_ABORT(mbc_slave_setup((void *)&comm_info));`

alisitsyn, 2023-03-24:
Hello @JimitD ,

Thank you for this report. Yes, the extended options are missing in the configuration structure. This was not addressed because the esp-modbus uses the legacy function prototypes from freemodbus. This is known issue and will be addressed in feature modbus update where the whole UART configuration structure will be propagated to modbus object constructor. For now, I would recommend you to use delayed init with call the function calls:
[uart_set_stop_bits](https://github.com/espressif/esp-idf/blob/master/components/driver/uart/include/driver/uart.h#L169) ,
[uart_set_word_length](https://github.com/espressif/esp-idf/blob/master/components/driver/uart/include/driver/uart.h#L135), in the application code after setup of modbus [here](https://github.com/espressif/esp-idf/blob/master/examples/protocols/modbus/serial/mb_master/main/master.c#L275). 
Note: even if you setup 2 stop bits but the peer sends the frame with one stop bit the ESP32 UART will not be able to recognize absence of one stop bit.

alisitsyn, 2024-01-17:
The updated serial options in v2.0.0_beta release: 
https://github.com/espressif/esp-modbus/discussions/45
https://github.com/espressif/esp-modbus/blob/release/v2.0/examples/serial/mb_serial_master/main/serial_master.c#L274

alisitsyn, 2024-04-19:
@JimitD , the documentation for v1.0.x will be updated to describe how to override the communication options:

[image]

alisitsyn, 2024-08-30:
Closed as solved.

closed 2024-08-30 by alisitsyn