espressif/esp-modbus #26: Kconfig FMB_TIMER_ISR_IN_IRAM removed but used

kossnikita, 2023-04-06:
After commit https://github.com/espressif/esp-idf/commit/ee104f8de2b98bb9961f201151ad72cc22385471 Kconfig option `FMB_TIMER_ISR_IN_IRAM` has been removed. But it still used in https://github.com/espressif/esp-modbus/blob/7d16826aef23729757ae21678c8bd20d385dd8ee/freemodbus/modbus/include/mbconfig.h#L174 
This causes the component to not work when flash is used.

alisitsyn, 2023-04-06:
Hi @kossnikita,

Thank you for this issue. I confirm that it was removed incompletely  in that commit from kconfig but still used in the code and can cause issues if it is not enabled but `CONFIG_FMB_TIMER_USE_ISR_DISPATCH_METHOD` is enabled in esp-idf => 5.0. However, this issues were not observed during test because the `sdkconfig.defaults` and test configs still have this option as enabled.

The fix for this issue should be as below:

[0001_fix_removed_option_CONFIG_FMB_TIMER_ISR_IN_IRAM.patch](https://github.com/espressif/esp-modbus/files/11172335/0001_fix_removed_option_CONFIG_FMB_TIMER_ISR_IN_IRAM.patch.log)

alisitsyn, 2023-06-21:
@kossnikita , 

The fix was merged in [v1.0.11](https://github.com/espressif/esp-modbus/releases/tag/v1.0.11).

alisitsyn, 2023-07-17:
The issue is closed. Feel free to reopen.

closed 2023-07-17 by alisitsyn