espressif/esp-modbus #28: Modbus RTU Master deadlock when sending with CONFIG_FMB_TIMER_PORT_ENABLED=y

gonzzor, 2023-04-20:
When running a Modbus RTU master with `CONFIG_FMB_TIMER_PORT_ENABLED=y` and the master only receives one or more 0 as response the stack ends up in a deadlock situation.

Tested with examples/protocols/modbus/serial/mb_master from IDF version 5.0.1 with ESP32-S3 Devkit.

Using the following sdkconfig changes
```
CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y
CONFIG_FMB_MASTER_TIMEOUT_MS_RESPOND=400
CONFIG_FMB_TIMER_PORT_ENABLED=y
CONFIG_FMB_TIMER_USE_ISR_DISPATCH_METHOD=y
```

The problem can be reproduces by running the following Python script on a host.
```python
#!/usr/bin/env python3
import serial
import datetime

port = serial.Serial('/dev/ttyUSB0', baudrate=115200, timeout=3/1000)

while True:
    data = port.read(1024)
    if data:
        print(datetime.datetime.now(), "data: ", data)
        port.write(b"\000")
        print("Write NUL")
```

The code is stuck in `eMBMasterWaitRequestFinish()` waiting for the transmission to complete.

Disabling CONFIG_FMB_TIMER_PORT_ENABLED solves the problem.

The bug seem to be at https://github.com/espressif/esp-modbus/blob/master/freemodbus/modbus/rtu/mbrtu_m.c#L290 because eRcvState is only set if ucByte != 0 but eSndState is changed to STATE_M_TX_IDLE.

When the T35 timer triggers later on nothing will happen in xMBMasterRTUTimerExpired() at https://github.com/espressif/esp-modbus/blob/master/freemodbus/modbus/rtu/mbrtu_m.c#L376 because eRcvState is STATE_M_RX_IDLE and eSndState is STATE_M_TX_IDLE. 

As timers are now disabled and nothing more will arrive on the UART the whole stack is currently deadlocked.

Something like this should solve the issue.
``` diff
diff --git a/freemodbus/modbus/rtu/mbrtu_m.c b/freemodbus/modbus/rtu/mbrtu_m.c
index bb44602..5fa585d 100644
--- a/freemodbus/modbus/rtu/mbrtu_m.c
+++ b/freemodbus/modbus/rtu/mbrtu_m.c
@@ -284,12 +284,12 @@ xMBMasterRTUReceiveFSM( void )
          * Disable timer of respond timeout and change the transmiter state to idle.
          */
         vMBMasterPortTimersDisable( );
-        eSndState = STATE_M_TX_IDLE;
 
         usMasterRcvBufferPos = 0;
         if( xStatus && ucByte ) {
             ucMasterRTURcvBuf[usMasterRcvBufferPos++] = ucByte;
             eRcvState = STATE_M_RX_RCV;
+            eSndState = STATE_M_TX_IDLE;
         }
 
         /* Enable t3.5 timers. */
```

Thanks.

alisitsyn, 2023-05-02:
Hello @gonzzor ,

Thank you for the issue. This applies to ASCII communication as well. This is confirmed as a bug and started to follow the formal process.

alisitsyn, 2023-06-21:
@gonzzor ,

Thank you for report of the issue. The fix was merged in [v1.0.11](https://github.com/espressif/esp-modbus/releases/tag/v1.0.11)

closed 2023-07-17 by alisitsyn