espressif/esp-modbus #30: Invalid register reading after error "E (92525) MB_PORT_COMMON: vMBMasterRunResRelease(195): Resource release failure"

kotyara12, 2023-05-19:
- IDE: Visual Studio Code with Platformio extension
- IDF version: framework-espidf @ 3.40403.0 (4.4.3)
- Chip: ESP32-WROOM-32U
- Operating System: Windows
- Port master configuration: UART1, 9600bps
________
My project uses several sensors on the RS485 bus and the ModbusRTU protocol.
Including a sensor for temperature, humidity and other soil parameters. These sensors have a variable number of registers, as it depends on its modification - from 1 to 4. Therefore, I cannot read the registers "in a row", but have to do it "one by one".
I use the mbc_master_send_request function to read the sensor registers:

```
  mb_param_request_t _request = {
     .slave_addr = _address,
     .command = cmd,
     .reg_start=reg,
     .reg_size = 1
   };
   return mbc_master_send_request(&_request, (void*)value);
```

The following code reads only those registers that are preconfigured:
```
esp_err_t err = ESP_OK;
  uint16_t bufH, bufT, bufC, bufP = 0;

  // Humidity (moisture)
  if ((_item2) && (err == ESP_OK)) {
    err = readModbusRegister(FUNCTION_CODE_STATUS_READ, REG_STATUS_HUMIDITY, &bufH);
  };
  // Temperature
  if ((_item1) && (err == ESP_OK)) {
    err = readModbusRegister(FUNCTION_CODE_STATUS_READ, REG_STATUS_TEMPERATURE, &bufT);
  };
  // Conductivity
  if ((_item3) && (err == ESP_OK)) {
    err = readModbusRegister(FUNCTION_CODE_STATUS_READ, REG_STATUS_CONDUCTIVITY, &bufC);
  };
  // PH
  if ((_item4) && (err == ESP_OK)) {
    err = readModbusRegister(FUNCTION_CODE_STATUS_READ, REG_STATUS_PH, &bufP);
  };
```
For a while everything works fine!
But after a random time, an error is thrown:
```
E (92525) MB_PORT_COMMON: vMBMasterRunResRelease(195): Resource release failure.
```
**After that, a mess begins on the RS485 bus - the registers can start to "shift", for example, I request temperature, and PH or humidity comes through the bus. And exactly the same behavior is observed on other bus devices.**

The boss will rip my head off

PS: Yes, I saw in the examples that mbc_master_send_request is not directly called anywhere, but a scheme of pre-configured parameters is used. But it's kind of a shame. To read a couple of registers to create a super-bulky structure....

alisitsyn, 2023-05-19:
Hi @kotyara12 ,

The freemodbus in esp-idf v4.4.3 component is not updated to recent version. In order to check the issue could you try to update your modbus component in the freemodbus folder? This component folder must be replaced in the following path: `C:\Users\USERNAME\.platformio\packages\framework-espidf\components\freemodbus`. Please update the whole folder to the component files located [here](https://github.com/alisitsyn/modbus_support/tree/modbus_support/fix_glitches_esp_modbus_master_rtu_ascii/modbus_master_rtu/components/espressif__esp-modbus). Let me know if you have issues with this. Thanks.

kotyara12, 2023-05-19:
Hi, @alisitsyn 

Thanks!
I've been browsing through the issues here and came across a suggestion to add
```
set(EXCLUDE_COMPONENTS freemodbus)
```
I guess this should update espmodbus to an up to date version? I'll check a little later.

Alas, I can't use ESP-IDF 5.0.1 yet, because the problem with a crash when calling localtime_r has not yet been fixed there

alisitsyn, 2023-05-19:
@kotyara12 ,

Yes, thank you for update. Yes, you can use the latest version of v1.0.10 and add the above line to the cmake file to exclude the freemodbus component and use the latest esp-modbus instead. In most cases this allows to fix this issue. However, the latest component may have some issues in some rare cases when your application intensively uses WIFI and network services. In order to check this approach you can update the component to experimental one which I included above.

kotyara12, 2023-05-28:
@alisitsyn 

Thank you very much, this seems to have helped. At least the device works stably for several days, there are no bases of errors in data transmission over the bus.

closed 2023-05-28 by kotyara12

alisitsyn, 2023-05-30:
@kotyara12 ,

Thank you for update. It is good that it works for your project. This update is temporary and the event tracking will be updated in new revision of esp-modbus. Let me know if you still have any issues.