espressif/esp-modbus #32: mbc_master_send_request() doesn't work properly

Asanga-Viraj, 2023-08-10:
I am using ESP32 RTU ModBus master. The reading registers are working fine. 
But, writing register doesn't work properly. I am using mbc_master_send_request() function to write register. This function writes value of previous call each time for a specific reg_start address. An example code is below.

> uint16_t temp = 156;
> mb_param_request_t req = {};
> req.command = 6;
> req.reg_start = 1182;
> req.reg_size = 1;
> mbc_master_send_request(&req, (void *)&temp);
>
> temp = 75;
> req.command = 6;
> req.reg_start = 1182;
> req.reg_size = 1;
> mbc_master_send_request(&req, (void *)&temp);     //still writes value 156
> ESP_LOGI(TAG, "value:%d", temp);   //can check the value from here. It is automatically updated to previous value.

This issue can overcome by calling `mbc_master_send_request(&req, (void *)&temp);` with zero value before writing actual value. It seems to be calling this function release resources allocated for reg_start address.

What should I do? Am I making invalid requests?

alisitsyn, 2023-08-22:
Hi @Asanga-Viraj,

Thank you for the issue. I think more information from you is required to reproduce this issue.
I just checked this functionality and can confirm this works as expected with esp-modbus v1.0.11 and latest esp-idf.

The application:
```
void app_main(void)
{
    // Initialization of device peripheral and objects
    ESP_ERROR_CHECK(master_init());
    vTaskDelay(10);
    
    // Write registers to predefined state
    uint16_t register_data = 0x1111;
    register_data = 0x3333;
    mb_param_request_t req = {};
    req.slave_addr = 1; // The slave address is required in the request to send this properly!!!
    req.command = 6;
    req.reg_start = 1;
    req.reg_size = 1;
    mbc_master_send_request(&req, (void *)&register_data);

    register_data = 0x4444;
    req.command = 6;
    req.reg_start = 2;
    req.reg_size = 1;
    mbc_master_send_request(&req, (void *)&register_data); //still writes value 156
    ESP_LOGI("TEST", "value:%x", register_data); //can check the value from here. It is automatically updated to previous value.
}
```

The log of test application:
```
I (869) MODBUS_MASTER: Modbus master stack initialized...
D (969) MB_PORT_COMMON: xMBMasterRunResTake:Take MB resource (500 ticks).
D (969) MB_PORT_COMMON: xMBMasterRunResTake:Take MB resource (500 ticks).
D (969) MB_PORT_COMMON: 325441:EV_MASTER_FRAME_TRANSMIT
D (969) POLL transmit buffer: 06 00 01 33 33 
D (979) MB_PORT_COMMON: eMBMasterRTUSend: Port enter critical.
D (979) MB_PORT_COMMON: eMBMasterRTUSend: Port exit critical
D (989) MB_PORT_COMMON: xMBMasterPortSerialSendRequest default
D (999) MB_PORT_COMMON: vMBMasterPortTimersRespondTimeoutEnable Respond enable timeout.
D (999) MB_MASTER_SERIAL: MB_TX_buffer sent: (9) bytes.
D (1009) MB_PORT_COMMON: vMBMasterRxSemaRelease:RX semaphore is free.
D (1009) MB_PORT_COMMON: xMBPortSerialWaitEvent, UART event: 1 
D (1019) MB_MASTER_SERIAL: MB_uart[2] event:
D (1029) MB_MASTER_SERIAL: uart rx break.
D (1029) MB_PORT_COMMON: 325441:EV_MASTER_FRAME_SENT
D (1029) MB_PORT_COMMON: xMBPortSerialWaitEvent, UART event: 0 
D (1039) MB_MASTER_SERIAL: MB_uart[2] event:
D (1039) MB_MASTER_SERIAL: Data event, len: 8.
D (1049) MB_MASTER_SERIAL: Received data: 9(bytes in buffer)
D (1049) MB_MASTER_SERIAL: Timeout occured, processed: 9 bytes
D (1059) POLL sent buffer: 06 00 01 33 33 
D (1059) MB_PORT_COMMON: 325441:EV_MASTER_FRAME_RECEIVED
D (1069) MB_PORT_COMMON:  xMBMasterPortSerialGetResponse default
D (1079) MB_PORT_COMMON: eMBMasterRTUReceive: Port enter critical.
D (1079) MB_PORT_COMMON: eMBMasterRTUReceive: Port exit critical
D (1089) MB_PORT_COMMON: 325441: Packet data received successfully (0).
D (1099) POLL receive buffer: 06 00 01 33 33 
D (1099) MB_PORT_COMMON: 325441:EV_MASTER_EXECUTE
D (1099) MB_PORT_COMMON: 325441:set event EV_ERROR_OK
D (1109) MB_PORT_COMMON: 325441:EV_MASTER_ERROR_PROCESS
D (1119) MB_PORT_COMMON: vMBMasterCBRequestSuccess: Callback request success.
D (1119) MB_PORT_COMMON: Transaction (325441), processing time(us) = 141985
D (1129) MB_PORT_COMMON: eMBMasterWaitRequestFinish: returned event = 0x80
D (1139) MB_PORT_COMMON: xMBMasterRunResTake:Take MB resource (500 ticks).
D (1139) MB_PORT_COMMON: xMBMasterRunResTake:Take MB resource (500 ticks).
D (1149) MB_PORT_COMMON: 507794:EV_MASTER_FRAME_TRANSMIT
D (1159) POLL transmit buffer: 06 00 02 44 44 
D (1159) MB_PORT_COMMON: eMBMasterRTUSend: Port enter critical.
D (1169) MB_PORT_COMMON: eMBMasterRTUSend: Port exit critical
D (1169) MB_PORT_COMMON: xMBMasterPortSerialSendRequest default
D (1179) MB_PORT_COMMON: vMBMasterPortTimersRespondTimeoutEnable Respond enable timeout.
D (1189) MB_MASTER_SERIAL: MB_TX_buffer sent: (9) bytes.
D (1189) MB_PORT_COMMON: xMBPortSerialWaitEvent, UART event: 1 
D (1199) MB_MASTER_SERIAL: MB_uart[2] event:
D (1199) MB_MASTER_SERIAL: uart rx break.
D (1209) MB_PORT_COMMON: 507794:EV_MASTER_FRAME_SENT
D (1209) POLL sent buffer: 06 00 02 44 44 
D (1229) MB_PORT_COMMON: xMBPortSerialWaitEvent, UART event: 0 
D (1229) MB_MASTER_SERIAL: MB_uart[2] event:
D (1229) MB_MASTER_SERIAL: Data event, len: 8.
D (1229) MB_MASTER_SERIAL: Received data: 9(bytes in buffer)
D (1239) MB_MASTER_SERIAL: Timeout occured, processed: 9 bytes
D (1239) MB_PORT_COMMON: 507794:EV_MASTER_FRAME_RECEIVED
D (1249) MB_PORT_COMMON:  xMBMasterPortSerialGetResponse default
D (1259) MB_PORT_COMMON: eMBMasterRTUReceive: Port enter critical.
D (1259) MB_PORT_COMMON: eMBMasterRTUReceive: Port exit critical
D (1269) MB_PORT_COMMON: 507794: Packet data received successfully (0).
D (1269) POLL receive buffer: 06 00 02 44 44 
D (1279) MB_PORT_COMMON: 507794:EV_MASTER_EXECUTE
D (1279) MB_PORT_COMMON: 507794:set event EV_ERROR_OK
D (1289) MB_PORT_COMMON: 507794:EV_MASTER_ERROR_PROCESS
D (1289) MB_PORT_COMMON: vMBMasterCBRequestSuccess: Callback request success.
D (1299) MB_PORT_COMMON: Transaction (507794), processing time(us) = 138328
D (1309) MB_PORT_COMMON: eMBMasterWaitRequestFinish: returned event = 0x80
I (1309) TEST: value:4444
```
host software log (RS485 interface):
```
Rx:001949-11:41:09.766-01 06 00 01 33 33 8C EF
Tx:001950-11:41:09.766-01 06 00 01 33 33 8C EF
Rx:001951-11:41:09.939-01 06 00 02 44 44 1B 39
Tx:001952-11:41:09.940-01 06 00 02 44 44 1B 39
```
If you are sure that it works incorrectly please send your code to reproduce this issue, log of your application with debug verbosity enabled in kconfig and log of the host application (RS485 bus log).

Thank you.

alisitsyn, 2023-09-04:
@Asanga-Viraj,

Any update on this issue? Could you please let me know which version of esp-idf and esp-modbus component you are using?

Asanga-Viraj, 2023-09-07:
Hi,

I am running complex system in ESP32 with idf 5.0.2 (release) and modbus 1.0.11 (latest) . I extracted what I did for ModBus write to replicate the issue in a single file. I couldn't replicate the issue. Tried many ways, I couldn't.

So, may be some fault in my side which I didn't recognized. Thank you for your help.

alisitsyn, 2023-09-08:
Hi @Asanga-Viraj,

Could you try to describe your complex system? For example: which active objects (tasks, interrupts) you have and how they dependent to each other and their priorities compare to Modbus priorities, Modbus settings in sdkconfig, etc. Do you have the tasks with priority >= 20 in your system.The modbus uses esp-timer and its task priority is 21, and higher priority tasks can apply to its functionality as well. Other components (wifi, bluethooth, lwip related) are also apply to the functionality of the whole system with their high priority tasks (prio up to 23). Does the issue exist only when you use the `mbc_master_send_request` to write the registers or the regular write API functions work the same way?
I think the active objects dependencies is the important point to be investigated on high level design stage.
Try to add the task with priority like something like ~20 into your extracted code and check issue again.

Thanks.

Asanga-Viraj, 2023-09-12:
Hi @alisitsyn ,

I will try to explain the system as much as I can.

- System consist of internet connectivity 3 interfaces such as WiFi, Ethernet and 4G PPP. Only one interface is running at a time.
- Once internet is connected, the system is running MQTT (not SSL) to publish data.
- Meanwhile RTU ModBus master is running.
      - baud 9600
      - parity none
      - Slave respond timeout = 300ms
      - Slave conversion delay = 200ms
      - Modbus serial task queue length = 20
      - Modbus port task stack size = 4096
      - Modbus serial task RX/TX buffer size = 256
      - Modbus port task priority = 10
      - Modbus task affinity = CPU0
      - Modbus controller notification timeout = 20
      - Modbus controller notification queue size = 20
      - Modbus controller stack size = 4096
      - Modbus stack event queue timeout  = 20ms
      - Modbus stack use timer for 3.5T symbol time measurement = unticked
      - Modbus timer uses ISR dispatch method = unticked

- ModBus reading and writing each 200ms
- Read data comparison tasks and other several tasks are running.
- two esp timers are running for data publishing and led indication.
- gpio interrupts are running for digital inputs.

Let me know any more information you may want.

Thanks.

alisitsyn, 2023-09-12:
Hi @Asanga-Viraj,

Thank you for update. Could you share the sdkconfig file as well? Your logging information can also help to improve your system.

> Read data comparison tasks and other several tasks are running.

Please include the tasks priority for most of your tasks including the tasks which do Modbus read data and comparison. This will help to find solutions for your case.

> ModBus reading and writing each 200ms

I would suggest to increase the `CONFIG_FMB_MASTER_TIMEOUT_MS_RESPOND` at least to 400 ms, even through your slave is able to process request during 200ms or less and increase the read/write period. The 200ms response timeout is to short response time for your baud rate and highly loaded system. How the modbus read/write is scheduled in your code?

> Modbus task affinity = CPU0

My suggestion is to change affinity setting to CPU1. This is required because your LWIP related services  (WIFI, ethernet, PPP) use high priority tasks and process data intensively. This allows to increase the reliability of Modbus data procesisng due to deterministic aspects of protocol. In this case it is recommended to move deterministic protocols to other CPU. This will significantly increase the reliability of data reading/writing in highly loaded system.

> Modbus timer uses ISR dispatch method = unticked

I recommend to set `CONFIG_FMB_TIMER_USE_ISR_DISPATCH_METHOD=y`. The esp-timer which is used by modbus by default use dispatch method `ESP_TIMER_TASK` when the above setting is disabled. This may cause the decreased accuracy for response timeout event triggering in case of heavily loaded system. If this setting is enabled, the timeout is dispatched from ISR.

Please try the suggestions above and report the results. Thank you.

Asanga-Viraj, 2023-09-17:
Hi,

Thank you for the support. I updated the values as requested and did testing. Still got the issue. It is not happening all the times, but randomly occurs. I updated the code to handle this mis ModBus writing.

As a information, I added sdkconfig file here.

[sdkconfig.zip](https://github.com/espressif/esp-modbus/files/12643910/sdkconfig.zip)

alisitsyn, 2023-09-29:
Hi @Asanga-Viraj ,

It is very strange. I need to reproduce this, but could not do it yet. Let me know if you have any update for this.

alisitsyn, 2025-05-07:
Sorry for long processing of this issue. It is supposed that these kind of issues are fixed in latest v1 and  v2  component releases. The appropriate fixes where merged.