espressif/esp-modbus #34: The maximum number of supported Modbus functions is too short

fj0207, 2023-08-21:
The maximum number of supported Modbus functions MB_FUNC_HANDLERS_MAX is actually set to 16 and cannot be changed by configuration variable. 
However Modbus specifications allow to define 19 user-defined functions in addition to the basic modbus functions. 

MB_FUNC_HANDLERS_MAX should be increase or at least configurable. Using SDK config.

alisitsyn, 2023-08-21:
Hi fj0207,

Thank you for the issue. The current stack supports only the standard commands. The change of the MB_FUNC_HANDLERS_MAX to configurable value will not allow you to add your custom command. The custom function handler should be registered appropriately with specific API call. This functionality is not supported in stack v1.x.x because it is hard to support and track all related user issues later. However, it is planned to add this functionality later.
You will have to manually add your command handler to xMasterFuncHandlers(xFuncHandlers) function table . You can raise a feature request or submit a patch if you have one handy.

fj0207, 2023-08-21:
Yes, indeed. 
I simply did it by calling the eMBRegisterCB from my code.  This function should be changed as public in modbus component to avoid adding declaration in user code.
Our modbus devices are using 9 user-defined functions with such modifications.

alisitsyn, 2023-08-21:
This sounds good to me. I don't think it is good solution to make the function public. The external API interface will not be changed in this version of stack but the internal freemodbus API designed to be isolated from user application. Please consider this change as a private update for now.