espressif/esp-modbus #37: Timer ISR dispatch mode configuration error

allard-potma, 2023-09-13:
Hi,
I noticed a small error in the timer isr dispatch mode configuration:
The kconfig file names the configuration option
`CONFIG_FMB_TIMER_USE_ISR_DISPATCH_METHOD`
In [esp-modbus](https://github.com/espressif/esp-modbus/tree/master)/[freemodbus](https://github.com/espressif/esp-modbus/tree/master/freemodbus)/[modbus](https://github.com/espressif/esp-modbus/tree/master/freemodbus/modbus)/[include](https://github.com/espressif/esp-modbus/tree/master/freemodbus/modbus/include)

/mbconfig.h

The define is renamed, but with the wrong name:
`CONFIG_FMB_TIMER_ISR_IN_IRAM`
This results in a timer ISR that is not placed into IRAM.

alisitsyn, 2023-09-13:
Hi, 

Ah, it seems the commits were squashed incorrectly. The issue is confirmed and will be fixed in coming release. Thanks you.

alisitsyn, 2023-09-29:
Merged in release v1.0.12, commit ea3f06cf2b56ec3839d76cf9844f46602b536223

closed 2023-10-10 by alisitsyn