espressif/esp-modbus #38: function vMBPortTimersDelay is missing

jamesarm97, 2023-10-20:
Just updated and my linker is complaining about vMBPortTimersDelay is missing. I see it referenced in mbport.h and mb.c. Was this removed and you have to provide it in your code now? It was at the end of port/porttimer.c

void vMBPortTimersDelay(USHORT usTimeOutMS)
{
vTaskDelay(usTimeOutMS / portTICK_PERIOD_MS);
}

alisitsyn, 2023-10-23:
Hello @jamesarm97 ,

Thank you for the issue. You are right and this is mistake. So, the `mbport.h` file contains the declaration of the function `vMBPortTimersDelay()` prototype. If the `MB_ASCII_TIMEOUT_WAIT_BEFORE_SEND_MS!=0 && comm mode is ASCII` the function is used otherwise this does not cause the linker issue. The issue is confirmed and will be updated later accordingly.

alisitsyn, 2024-01-03:
The fix is merged with commit ID 94e9664aaa552718f1c2d76ef3490edf755fc7cd.

closed 2024-01-03 by alisitsyn