espressif/esp-modbus #46: modbus slave stack

kush930, 2023-12-21:
Hi all,

`I want to build esp-modbus source on my machine x86 linux os , I cloned successfully master branch from link-` https://github.com/espressif/esp-modbus.git ,  `but here i am not getting proper idea how to compile and install  the source.`

**if anyone knows, can anyone guide me or suggest me, how can i build source.**

T&R
Kush Kumar

alisitsyn, 2023-12-21:
Hi Kush Kumar,

In order to build the esp-modbus examples you need to install esp-idf framework and build tools and utils. The `esp-modbus` component will be installed automatically by component manager. Please follow the steps below to accomplish this:

1. Follow the [guide to install esp-idf and tools](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html#get-started-linux-macos-first-steps) and start simple [helloworld application](https://github.com/espressif/esp-idf/tree/master/examples/get-started/hello_world)  from esp-idf framework. All information on how to do this is included in the guide link above. 
3. Prepare as an example the [ESP32-DevKitC](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-devkitc.html), but other boards can be used as well (two boards for master and slave example). Just open the terminal application, connect your board to usb port of the computer and type commands:
```
cd $IDF_PATH/examples/get-started/hello_world
idf.py set-target esp32
idf.py build
idf.py flash monitor
``` 
Make sure the application is built successfully and last step flashes the application to the board.
4. Prepare esp32 boards that will be used to check the modbus communication. In order to use RTU communication you need to buy: 
- minimal you need two RS485-TTL adapters for your boards. [An example](https://www.amazon.com/WWZMDiB-3-3V-5-5V-Conversion-Hardware-Automatic/dp/B0BKG7SC54/ref=sr_1_2_sspa?crid=2RXZIPM1H83U8&keywords=ttl+to+rs485+max483&qid=1703153192&sprefix=ttl+to+rs485+max483+%2Caps%2C190&sr=8-2-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&psc=1). [Other similar variant](https://www.amazon.com/YWBL-WH-RS-485-Converter-Adapter-Raspberry/dp/B07S1BN6ZK/ref=sr_1_3?crid=2RXZIPM1H83U8&keywords=ttl+to+rs485+max483&qid=1703153423&sprefix=ttl+to+rs485+max483+%2Caps%2C190&sr=8-3) with DE/~RE pins. 
- [An USB-to-RS485 adapter](https://www.amazon.com/NOYITO-Converter-Interface-Dual-Function-Dual-Protection/dp/B07D529B1X/ref=sr_1_8?crid=2RXZIPM1H83U8&keywords=ttl+to+rs485+max483&qid=1703153423&sprefix=ttl+to+rs485+max483+%2Caps%2C190&sr=8-8) to be able to check communication over RS485 with your PC computer later.
5. Build the official [esp-idf Modbus master example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/modbus/serial/mb_master). Go to the folder `$IDF_PATH/examples/protocols/modbus/serial/mb_master` and follow the readme.md file which describes how to connect the RS485 adapter to your ESP32 board. Then build and flash the application in your terminal
```
$IDF_PATH/examples/protocols/modbus/serial/mb_master
idf.py menuconfig
```
[image], set the options as on the picture and then build the application, connect your ESP32 board to USB port of your PC:
```
$IDF_PATH/examples/protocols/modbus/serial/mb_master
idf.py build
idf.py flash monitor
```
6. Follow the steps above for the Modbus slave project: `$IDF_PATH/examples/protocols/modbus/serial/mb_slave`.
The configuration options `idf.py menuconfig`:
[image]
```
$IDF_PATH/examples/protocols/modbus/serial/mb_slave
idf.py build 
idf.py flash monitor
```
7. Connect Master board RS485 adapter to Slave board RS485 adapter signals A->A, B->B.
8. Reset Slave board and then Master board and make sure they can communicate each other over RS485. Make sure you connected the pins of ESP32 board to RS485 adapters correctly as described in `readme.md` file of master and slave project accordingly. 
slave log: 
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7172
load:0x40078000,len:15532
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:3904
entry 0x40080640
I (28) boot: ESP-IDF v5.3-dev-1123-g854d2646908 2nd stage bootloader
I (28) boot: compile time Dec 21 2023 11:50:51
I (30) boot: Multicore bootloader
I (34) boot: chip revision: v1.0
I (38) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 2MB
I (52) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (95) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0d388h ( 54152) map
I (122) esp_image: segment 1: paddr=0001d3b0 vaddr=3ffb0000 size=0248ch (  9356) load
I (125) esp_image: segment 2: paddr=0001f844 vaddr=40080000 size=007d4h (  2004) load
I (128) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=18d3ch (101692) map
I (171) esp_image: segment 4: paddr=00038d64 vaddr=400807d4 size=0ea08h ( 59912) load
I (202) boot: Loaded app from partition at offset 0x10000
[... 107 lines cut ...]
I (1396) MASTER_TEST: Characteristic #9 DiscreteInpP1 (on/off) value = OFF (0x15) read successful.
I (1906) MASTER_TEST: Characteristic #0 Data_channel_0 (Volts) value = 1.120000 (0x3f8f5c29) read successful.
I (1906) MASTER_TEST: Characteristic #1 Humidity_1 (%rH) value = 1.740000 (0x3fdeb853) read successful.
I (1916) MASTER_TEST: Characteristic #2 Temperature_1 (C) value = 2.340000 (0x4015c28f) read successful.
[... cut ...]
I (2556) MASTER_TEST: Characteristic #6 Test_regs (__) value = (0xaaaaaaaa) read successful.
I (2566) MASTER_TEST: Characteristic #7 RelayP1 (on/off) value = OFF (0x15) read successful.
I (2566) MASTER_TEST: Characteristic #8 RelayP2 (on/off) value = OFF (0x2a) read successful.
I (2576) MASTER_TEST: Characteristic #9 DiscreteInpP1 (on/off) value = OFF (0x15) read successful.
I (3086) MASTER_TEST: Characteristic #0 Data_channel_0 (Volts) value = 1.120000 (0x3f8f5c29) read successful.
```

Let me know if you have issues with this.

alisitsyn, 2024-01-02:
@kush930 ,
Do you have any update on this? If the issue is solved, please close the ticket.
Thanks.

alisitsyn, 2024-01-24:
The issue is closed. Feel free to reopen.

closed 2024-01-24 by alisitsyn