espressif/esp-modbus #48: Write Single Register Configuration

dmglogowskiOS, 2024-01-17:
Hello!

We are running into issues trying to understand how to use this component properly, specifically in regards on how to write to a single holding register.

Our Slave Device does not currently support writing multiple registers using function code `0x10` as this functionality wasn't yet needed as up till now we used the function code `0x06` to write to a single Holding Register. 

We know that we could manually write the request but this is more error prone in our opinion. 

Therefore the question, is it possible to configure a member of our `mb_parameter_descriptor_t[]` to use Function code `0x06` instead of `0x10`?

alisitsyn, 2024-01-22:
Hi,

Custom implementation is also possible. I also have the ideas on how to accomplish this but currently I am busy with some other things. Please feel free to share your ideas. I will try to review as soon as possible from my side. It is important for me to provide more convenient way for users related to this feature.    

Thanks.

dmglogowskiOS, 2024-01-22:
Hi,

I will get into working on that either later today or sometime later this week. Either way, no problem.

alisitsyn, 2024-01-22:
One important comment: I have an requirement for the project is that the external API interface for v1.0.x of esp-modbus should not be changed. This includes the format of Data dictionary. The changes of interface for v1.0.x will not be applied. This is why I did not changed this directly but added the workaround for this as an additional function. However, the changes you are working on can be implemented in [esp-modbus v2.0.0-beta](https://github.com/espressif/esp-modbus/discussions/45).

dmglogowskiOS, 2024-01-22:
Entirely understandable, my approach would likely end up adding to the Data Dictionary and thus causing this requirement to break.

However I am of the opinion that sometimes you gotta break compatibility, document it and be fine with it.

alisitsyn, 2024-01-26:
> However I am of the opinion that sometimes you gotta break compatibility, document it and be fine with it.

I have got your opinion and will try to address this when possible.

dmglogowskiOS, 2024-01-26:
I think I actually found a way that wouldn't break compatibility but still allow commands to be configurable when wanted while the current behavior stays default.

DomiMartinGlogi, 2024-01-26:
[PR is opened regarding this](https://github.com/espressif/esp-modbus/pull/50)

alisitsyn, 2024-01-26:
The PR will not be merged due to the breaking changes. For v1.0.x I recommend to use the existing compatibility way of sending the custom requests with command 0x05 which works just fine. See the documentation and examples mentioned above.

DomiMartinGlogi, 2024-01-27:
As far as I am aware the changes shouldn't be breaking, however I would love to get more detailed information as to why you think they are.

Either way, implementing these changes should be, imo, done for the next major version.

alisitsyn, 2024-05-10:
@ DomiMartinGlogi ,

Sorry for even very late response. Unfortunately your code changes the interface and will cause the maintaining issues later. I have other idea to implement this using compatible way. I will address this when possible.

alisitsyn, 2025-04-07:
@DomiMartinGlogi , the support for custom commands is merged with commit [cbe96056f2bcb44d4580355126b7255af332c3ba](https://github.com/espressif/esp-modbus/commit/cbe96056f2bcb44d4580355126b7255af332c3ba). This allows to add or override the custom handler for a function or get existing one for each Slave or Master object. If you have time, please take a look and verify this functionality. This will be available in next release. The functionality to use the custom commands from data dictionary will be merged a bit later.

closed 2025-06-30 by alisitsyn