espressif/esp-modbus #49: ESP-Modbus v2.0.0-beta Modubs TCP Slave crash restart

greentd, 2024-01-18:
I need to run both Modbus RTU and Modbus TCP Slave on ESP32S3, I found ESP-Modbus v2.0.0-beta can meet my needs,
But when I tested examples\tcp\mb_tcp_slave again, there was this statement in the function static mb_event_group_t mbc_tcp_slave_check_event(void *ctx, mb_event_group_t group)
BaseType_t status = xEventGroupWaitBits(mbs_opts->event_group_handle, (BaseType_t)group, pdTRUE, pdFALSE, portMAX_DELAY); will cause ESP32S3 to crash and restart.

The error prompt is as follows

D (5852) mb_object.slave: created object mbs_tcp@0x3fcb1cc8
W (5852) mb_port.timer: mbs_tcp@0x3fcb1cc8, start timer (1000000).
W (5862) mb_object.slave: 0x3fcb1cc8:EV_READY
I (5912) SLAVE_TEST: Modbus slave stack initialized.
I (5912) SLAVE_TEST: Start modbus test...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x42002dbb
0x42002dbb: panic_handler at D:/esp32/esp-idf/components/esp_system/port/panic_handler.c:146

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x178c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xcb8
load:0x403cc700,len:0x2d98
entry 0x403c9914
I (31) boot: ESP-IDF v5.3-dev-1353-gb3f7e2c8a4 2nd stage bootloader
I (31) boot: compile time Jan 18 2024 15:37:04

How to solve this?

alisitsyn, 2024-01-18:
Hi @greentd,

Thank you for the issue. This happens because the port for Modbus TCP Slave is not implemented yet in v2.0.0. There are placeholders instead of real implementation. The implementation will be updated after completion of some other tasks. I will inform you as soon as it is ready.

greentd, 2024-01-18:
> Hi @greentd,
> 
> Thank you for the issue. This happens because the port for Modbus TCP Slave is not implemented yet in v2.0.0. There are placeholders instead of real implementation. The implementation will be updated after completion of some other tasks. I will inform you as soon as it is ready.

Looking forward to your updates!

fink-at-trmc-dk, 2024-02-27:
Also waiting for this update :-)

alisitsyn, 2024-09-04:
Hi @greentd,  @fink-at-trmc-dk,
The `esp-modbus v2.0.0-beta.1` component is released.
The updated version supports the multi-instance TCP Slave and Master with its own communication options. The pre-release esp-modbus component version [2.0.0-beta.1](https://components.espressif.com/components/espressif/esp-modbus/versions/2.0.0-beta.1) can be selected by `idf_component.yml` in main folder like below:

``` yml
dependencies:
  idf:
     version: ">=4.1.0"
  espressif/esp-modbus:
    version: "^2.0.0-beta"
  espressif/mdns:
    version: "^1.0.0"
    rules:
      - if: "idf_version >=5.0"
  mb_example_common:
    path: "../mb_example_common"
  protocol_examples_common:
    path: ${IDF_PATH}/examples/common_components/protocol_examples_common
```

alisitsyn, 2024-09-04:
The issue is closed, feel free to reopen if you have any issues.

closed 2024-09-04 by alisitsyn