espressif/esp-modbus #52: esp-modbus init MB_RETURN_ON_FALSE compilation exception

yel-best, 2024-01-30:
The following problem occurred when I compiled, I specified esp-modbus version 1.0.13

```C
// FreeRTOS
#include <freertos/FreeRTOS.h>
#include <freertos/timers.h>
#include <freertos/semphr.h>

#include "string.h"

#include "app_power_modbus.h"
#include "mbcontroller.h"
#include "modbus_params.h" // for modbus parameters structures
#include "sdkconfig.h"

// Modbus master initialization
static esp_err_t master_init(void)
{
    // Initialize and start Modbus controller
    mb_communication_info_t comm = {
        .port = MB_PORT_NUM,
#if CONFIG_MB_COMM_MODE_ASCII
        .mode = MB_MODE_ASCII,
#elif CONFIG_MB_COMM_MODE_RTU
        .mode = MB_MODE_RTU,
#endif
        .baudrate = MB_UART_SPEED,
        .parity = MB_PARITY_NONE
    };
    void *master_handler = NULL;

    esp_err_t err = mbc_master_init(MB_PORT_SERIAL_MASTER, &master_handler);
    MB_RETURN_ON_FALSE((master_handler != NULL), ESP_ERR_INVALID_STATE, TAG,
                       "mb controller initialization fail.");
    MB_RETURN_ON_FALSE((err == ESP_OK), ESP_ERR_INVALID_STATE, TAG,
                       "mb controller initialization fail, returns(0x%x).",
                       (uint32_t)err);
    err = mbc_master_setup((void *)&comm);
    MB_RETURN_ON_FALSE((err == ESP_OK), ESP_ERR_INVALID_STATE, TAG,
                       "mb controller setup fail, returns(0x%x).",
                       (uint32_t)err);

    // Set UART pin numbers
    err = uart_set_pin(MB_PORT_NUM, MB_UART_TXD, MB_UART_RXD, CONFIG_MB_UART_RTS, UART_PIN_NO_CHANGE);
    MB_RETURN_ON_FALSE((err == ESP_OK), ESP_ERR_INVALID_STATE, TAG,
                       "mb serial set pin failure, uart_set_pin() returned (0x%x).", (uint32_t)err);

    err = mbc_master_start();
    MB_RETURN_ON_FALSE((err == ESP_OK), ESP_ERR_INVALID_STATE, TAG,
                       "mb controller start fail, returns(0x%x).",
                       (uint32_t)err);

    // Set driver mode to Half Duplex
    err = uart_set_mode(MB_PORT_NUM, UART_MODE_RS485_HALF_DUPLEX);
    MB_RETURN_ON_FALSE((err == ESP_OK), ESP_ERR_INVALID_STATE, TAG,
                       "mb serial set mode failure, uart_set_mode() returned (0x%x).", (uint32_t)err);

    vTaskDelay(5);
    err = mbc_master_set_descriptor(&device_parameters[0], num_device_parameters);
    MB_RETURN_ON_FALSE((err == ESP_OK), ESP_ERR_INVALID_STATE, TAG,
                       "mb controller set descriptor fail, returns(0x%x).",
                       (uint32_t)err);
    ESP_LOGI(TAG, "Modbus master stack initialized...");
    return err;
}

```

```C
In file included from E:/MyProject/IoT/esp/esp-idf/components/esp_common/include/esp_check.h:9,
                 from E:/MyProject/IoT/PlatformIO-ESP/esp-idf-mesh-lite/mesh-lite-test/managed_components/espressif__esp-modbus/freemodbus/common/include/esp_modbus_common.h:18,
                 from E:/MyProject/IoT/PlatformIO-ESP/esp-idf-mesh-lite/mesh-lite-test/managed_components/espressif__esp-modbus/freemodbus/common/include/esp_modbus_master.h:13,
                 from E:/MyProject/IoT/PlatformIO-ESP/esp-idf-mesh-lite/mesh-lite-test/managed_components/espressif__esp-modbus/freemodbus/common/include/mbcontroller.h:20,
                 from E:/MyProject/IoT/PlatformIO-ESP/esp-idf-mesh-lite/mesh-lite-test/main/app_power_modbus.c:10:
E:/MyProject/IoT/PlatformIO-ESP/esp-idf-mesh-lite/mesh-lite-test/main/app_power_modbus.c: In function 'master_init':
E:/MyProject/IoT/esp/esp-idf/components/log/include/esp_log.h:265:27: error: format '%x' expects argument of type 'unsigned int', but argument 8 has type 'long unsigned int' [-Werror=format=]
  265 | #define LOG_COLOR(COLOR)  "\033[0;" COLOR "m"

E:/MyProject/IoT/esp/esp-idf/components/log/include/esp_log.h:340:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
  340 | #define ESP_LOGE( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR,   tag, format, ##__VA_ARGS__)
      |                                      ^~~~~~~~~~~~~~~~~~~
E:/MyProject/IoT/esp/esp-idf/components/esp_common/include/esp_check.h:264:13: note: in expansion of macro 'ESP_LOGE'
  264 |             ESP_LOGE(log_tag, "%s(%d): " format, __FUNCTION__, __LINE__, ##__VA_ARGS__);        \
      |             ^~~~~~~~
E:/MyProject/IoT/PlatformIO-ESP/esp-idf-mesh-lite/mesh-lite-test/managed_components/espressif__esp-modbus/freemodbus/common/include/esp_modbus_common.h:21:59: note: in expansion of macro 'ESP_RETURN_ON_FALSE'       
   21 | #define MB_RETURN_ON_FALSE(a, err_code, tag, format, ...) ESP_RETURN_ON_FALSE(a, err_code, tag, format __VA_OPT__(,) __VA_ARGS__)
      |                                                           ^~~~~~~~~~~~~~~~~~~
E:/MyProject/IoT/PlatformIO-ESP/esp-idf-mesh-lite/mesh-lite-test/main/app_power_modbus.c:208:5: note: in expansion of macro 'MB_RETURN_ON_FALSE'
  208 |     MB_RETURN_ON_FALSE((err == ESP_OK), ESP_ERR_INVALID_STATE, TAG,
      |     ^~~~~~~~~~~~~~~~~~
cc1.exe: some warnings being treated as errors
ninja: build stopped: subcommand failed.

```

But when I comment out the code about **MB_RETURN_ON_FALSE**, it compiles normally. Is it because of compatibility issues?

[image]

alisitsyn, 2024-01-30:
Please do not remove the above macro. You can just add the line below into CMakeFiles.txt file in your `main` project folder.
```
target_compile_options(${COMPONENT_LIB} PRIVATE "-Wno-format")
```
The issue happens because the you need correct type conversion in the format messages. The `%x` corresponds to `int` but the operand is converted to `uint32_t`. These lines come from some previous version of modbus and need to be fixed of just apply the change above to disable errors related to format checking.

yel-best, 2024-01-30:
> Please do not remove the above macro. You can just add the line below into CMakeFiles.txt file in your `main` project folder.
> 
> ```
> target_compile_options(${COMPONENT_LIB} PRIVATE "-Wno-format")
> ```
> 
> The issue happens because the you need correct type conversion in the format messages. The `%x` corresponds to `int` but the operand is converted to `uint32_t` (change to int, solves this). These lines come from some previous version of modbus and need to be fixed of just apply the change above to disable errors related to format checking.

Thank you for your answer. According to your guidance, I added this code to the path. The effect is as shown in the screenshot. However, there will be such a prompt when I compile. Did I add it correctly? 

After adding it, my compilation still failed. I restored all the changes and still couldn't compile. When I commented out the new sentence, I could compile normally. Why?

**Using idf version 5.1.2**

CMake Error at CMakeLists.txt:5 (target_compile_options):
  Cannot specify compile options for target "PRIVATE" which is not built by
  this project.

[image]

[image]

[image]

```C 
-- Configuring incomplete, errors occurred!
See also "E:/MyProject/IoT/PlatformIO-ESP/esp-idf-mesh-lite/mesh-lite-test/build/CMakeFiles/CMakeOutput.log".
FAILED: build.ninja 
E:\MyProject\IoT\esp\Tools\.espressif\tools\cmake\3.24.0\bin\cmake.exe --regenerate-during-build -SE:\MyProject\IoT\PlatformIO-ESP\esp-idf-mesh-lite\mesh-lite-test -BE:\MyProject\IoT\PlatformIO-ESP\esp-idf-mesh-lite\mesh-lite-test\build
ninja: error: rebuilding 'build.ninja': subcommand failed
```

alisitsyn, 2024-01-30:
As I said above the line needs to be added at the end of the file CMakeFiles.txt located in the `main` folder of your project.

yel-best, 2024-01-30:
> As I said above the line needs to be added at the end of the file CMakeFiles.txt located in the `main` folder of your project.

Yes you are right, it was my problem, sorry, I realized my mistake after I sent the query and it is valid now

yel-best, 2024-01-30:
I defined a custom component named `app_modbus`. The content of my `CMakeLists.txt `in the component is as follows, and I referenced the modbus_params.h file. This component implements all the modbus functions I need, but when I compile, some problems will appear. The error means that the `holding_reg_params, input_reg_params, coil_reg_params and discrete_reg_params` I defined were not found. However, if I add the content defined in `modbus_params.h` directly to the `app_modbus.c` file, this problem will not occur without specifying where the problem occurred. , is it related to the idf compilation chain? Please give me some help, thank you very much

```C
idf_component_register(SRCS "app_modbus.c"
                    INCLUDE_DIRS "include"
                    REQUIRES esp-modbus)
```

[image]

[image]

alisitsyn, 2024-01-31:
These structures for parameters are defined [here](https://github.com/espressif/esp-idf/tree/master/examples/protocols/modbus/mb_example_common). Add the path for this component to your CMakeLists.txt ( other variant put it like here to the [manifest](https://github.com/espressif/esp-idf/blob/master/examples/protocols/modbus/serial/mb_master/main/idf_component.yml#L5)) or copy this component to your project `components` folder.

alisitsyn, 2024-02-05:
@yel-best,

Could you please update the status of the issue? Did you solve the issue?

yel-best, 2024-03-04:
> @yel-best,
> 
> Could you please update the status of the issue? Did you solve the issue? Can this ticket be closed?

yes, what you said is correct. It is a problem with my code. You can close this issue. thanks.

closed 2024-03-04 by yel-best