espressif/esp-modbus #53: multi UART conflict

ss1k, 2024-02-20:
The project I'm working on right now, uses UART0 for NB-IOT communication, and UART1 for modbus. The problem occurs when setting up modbus, when it enters critical, the modbus's log info will be received by UART0. Following are those wrongly received message.
```
'D (59601) MB_PORT_COMMON: vMBPortSetMode: Port enter critical.[0m  
D (59601) MB_PORT_COMMON: vMBPortSetMode: Port exit critical[0m
D (59601) MB_PORT_COMMON: eMBMasterRTUInit: Port enter critical.[0m
D (59611) intr_alloc: Connected src 22 to int 9 (cpu 0)[0m
D (59611) MB_PORT_COMMON: eMBMasterRTUInit: Port exit critical[0m
D (59621) MB_PORT_COMMON: eMBMasterRTUStart: Port enter critical.[0m
D (59631) MB_PORT_COD (59631) DBG: RTU timer, init FSM state.[0m
D (59631) MBM_TIMER: Timer mode: (0) triggered[0m
MMON: eMBMasterRTUStart: Port exit critical[0m
ERROR

ERROR

ERROR

ERROR

ERROR

ERROR

ERROR

ERROR

ERROR
D (59691) MB_PORT_COMMON: eMBMasterRTUSend: Port enter critical.[0m
D (59691) MB_PORT_COMMON: eMBMasterRTUSend: Port exit critical[0m
D (59701) MB_PORT_COMMON: eMBMasterRTUReceive: Port enter critical.[0m
D (59701) MB_PORT_COMMON: eMBMasterRTUReceive: Port exit critical[0m
D (59701) MB_PORT_COMMON: eMBMasterRTUStop: Port enter critical.[0m
D (59711) MB_PORT_COMMON: eMBMasterRTUStop: Port exit critical[0m
D (59711) MB_PORT_COMMON: vMBPortSetMode: Port enter critical.[0m
D (59721) MB_PORT_COMMON: vMBPortSetMode: Port exit critical[0m
ERROR

ERROR

ERROR

ERROR

ERROR

ERROR

ERROR

ERROR
'
````

```
static _lock_t s_port_lock;
inline void
vMBPortEnterCritical(void)
{
    _lock_acquire(&s_port_lock);
}

inline void
vMBPortExitCritical(void)
{
    _lock_release(&s_port_lock);
}
```

I tried to assign s_port_lock to 1, since _lock_t is an Interger essentially, but it doesn't work. So 
I just commented out enter/exit critical and changed ESP_EARLY_LOGD into ESP_LOGD, most of the wrong message can be handled properly, but still 
```
'D (9799) intr_alloc: Connected src 22 to int 9 (cpu 0)[0m

ERROR'
```
this line of modbus information be received by NB-IOT.

Is there a way to change value of s_port_lock? or what are the other solutions to address my problem?

alisitsyn, 2024-02-21:
Hi @ss1k,

What kind of board and chip are used in your project? I guess you configured your ports incorrectly, NB_IOT and console use the same port = UART0. The default settings for console is UART0 and this is the port where the modbus and other software send the log messages.

[image]

ss1k, 2024-02-21:
> Hi @ss1k,
> 
> What kind of board and chip are used in your project? I guess you configured your ports incorrectly, NB_IOT and console use the same port = UART0. The default settings for console is UART0 and this is the port where the modbus and other software send the log messages. `idf.py menuconfig`: ![image](https://private-user-images.githubusercontent.com/37483886/306550605-630b6465-a974-4999-979c-df4268fd6bfa.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg1MDc0OTgsIm5iZiI6MTcwODUwNzE5OCwicGF0aCI6Ii8zNzQ4Mzg4Ni8zMDY1NTA2MDUtNjMwYjY0NjUtYTk3NC00OTk5LTk3OWMtZGY0MjY4ZmQ2YmZhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjIxVDA5MTk1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRkM2JjZWI0YTAwYzM5ZDlhNmRhNmFmYTA0OTBiODc5YmQ3MzYwMWU2MWE0M2I4ZTkxNjdlZDhmY2U2NmE3YTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.6dBZv4xEk6FySi8knY5jjf4C8l7kw7T6fIBL7HizQm8)

Hi @alisitsyn, I use a self-designed board and the chip is esp32c3, we solder the UART0 and NB_IOT module together, in this case, we can only use UART1 for modbus. The console port is UART0, and we can see the message from NB_IOT side, at the same time, the AT receiving message task of NB_IOT is running, that's why I don't wanna modbus's log message to be sent to UART0. Is there a way to change the UART of modbus? Thanks

alisitsyn, 2024-02-21:
> Is there a way to change the UART of modbus?

it is possible to change the working port of modbus but it is impossible to change the logging port for modbus. It is also possible to disable some messages of modbus but the component uses the unique tag for each module. I think the better way to change the logging behavior of the component is to move the component to `your_project/components` folder and redefine the logging macros to some custom redefined versions. In this case you will be able to control the logging of Modbus as you need.

[logging](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/log.html#master-logging-level)

ss1k, 2024-03-04:
> > Is there a way to change the UART of modbus?
> 
> it is possible to change the working port of modbus but it is impossible to change the logging port for modbus. It is also possible to disable some messages of modbus but the component uses the unique tag for each module. I think the better way is to change the logging behavior of the component moving it to `your_project/components` folder and redefine the logging macros to some custom redefined versions. In this case you will be able to control the logging of Modbus as you need.
> 
> [logging](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/log.html#master-logging-level)
> 
> > Is there a way to change value of s_port_lock? or what are the other solutions to address my problem?
> 
> The `s_port_lock` is the log for critical section of modbus port functionality. I don't think you need to change this. You can just redefine the log output macro as you need, the lock functions itself should not output anything into port.
> 
> > 'D (9799) intr_alloc: Connected src 22 to int 9 (cpu 0)[0m
> 
> This comes from interrupt allocation module and can be disabled: esp_log_level_set("intr_alloc", ESP_LOG_NONE);
> 
> Please also consider to disable the standard logging output functionality in this case and use your UART0 for NB_IOT only: * "idf.py menuconfig", go to "Bootloader config" and set "Bootloader log verbosity" to "No output" to disable bootloader output * "idf.py menuconfig", go to "Component config", "Log output", set "Default log verbosity" to "No output" to disable application output; Would this work for your project or you need some custom logging behavior also? Please let me know if this helps. Thanks.

Thank you so much for your help, we will change the logging behavior and move the modbus component under our project directory, I will let you know how's the result soon!

alisitsyn, 2024-09-04:
@ss1k,

How is the result on your side? I will close the issue due to inactivity. Feel free to reopen if you have any new update. 
Thanks.

closed 2024-09-04 by alisitsyn