espressif/esp-modbus #54: why support uart_hw_flowcontrol_t is not supported

yel-best, 2024-03-04:
Hi,

According to my understanding, it seems that esp-modbus does not support flow control?
Hardware flow control is not supported, and software flow control is not known to be supported.

What is the reason?

[image]

alisitsyn, 2024-04-02:
Hi @yel-best,

There are similar issues for this, one is answered [here](https://github.com/espressif/esp-modbus/issues/25). This stack supports the standard communication options for RTU and ASCII as per Modbus spec:

RTU: 
1 start bit
8 data bits, least significant bit sent first
1 bit for even/odd parity; no bit for no parity
1 stop bit if parity is used; 2 bits if no parity

ASCII:
1 start bit
7 data bits, least significant bit sent first
1 bit for even/odd parity; no bit for no parity
1 stop bit if parity is used; 2 bits if no parity

The API will not be changed here. However, If you need to use some custom serial options in your application you are free to do this. This can be accomplished by overriding  the serial settings after call to `mbc_master_setup()`. See the related link above.

yel-best, 2024-04-17:
> Hi @yel-best,
>
> There are similar issues for this, one is answered [here](https://github.com/espressif/esp-modbus/issues/25). This stack supports the standard communication options for RTU and ASCII [...]

oh, ok, thanks for your advice 😄

alisitsyn, 2024-04-18:
@yel-best, the issue will be closed.

closed 2024-04-18 by alisitsyn

alisitsyn, 2024-04-19:
Update: the documentation will be updated a per your issue as below:
[image]