espressif/esp-modbus #55: Add library on platformio

mattia424, 2024-03-22:
Bye, 
Is there a simple way to load this library into the platformio environment?
I have tried all ways but from the main.c I can't reach the header files

alisitsyn, 2024-03-25:
@mattia424 ,

The esp-modbus can be used with platforio environment. In order to accomplish this you need:

1. Download the latest library from [v1.013]().
2. Extract it to the path `c:\Users\youruser\.platformio\packages\framework-espidf\components\` (Win path! should be adopted on linux) or to the 'your_proj_root\components' folder.
3. Create your project in platformio. In your root folder CMakeFiles.txt file add the lines as below:
```
cmake_minimum_required(VERSION 3.16.0)
set(EXCLUDE_COMPONENTS freemodbus) #to exclude the old freemodbus component which is part of platformio
# Include parameters from common modbus folder (add the common parameters folder in the root of the project, if you need them
set(MB_PARAMS_DIR "${CMAKE_CURRENT_SOURCE_DIR}/mb_example_common")
list(APPEND EXTRA_COMPONENT_DIRS "${MB_PARAMS_DIR}")
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(mbm)
```
4. In your `pio_proj_root\src` in the `main.c` you can include the  
`modbus_params.h`, `mbcontroller.h` and implement your modbus implementation using the stack API.
5. Compile the project in platformio CLI.

[CMakeLists.txt](https://github.com/espressif/esp-modbus/files/14742542/CMakeLists.md)

mattia424, 2024-03-27:
Perfect this solved my problem, I had missed the fact that it is a component and not just a library to be included

alisitsyn, 2024-03-28:
Thank you for the feedback. The issue will be closed, feel free to reopen.

closed 2024-03-28 by alisitsyn