espressif/esp-modbus #60: mbc_master_send_request - Exception Codes

dmglogowskiOS, 2024-05-03:
When `mbc_master_send_request` returns an `ESP_FAIL` to the caller, the documentation states that an exception or other failure occurred. However, it is unclear how to determine what might be the case, since:
1. ESP_FAIL indicates that either an exception occurred OR that there was another failure
2.  It is not clear if the modbus exception codes are accessible, when an exception occurred. 

Clarifying information, and adding to the documentation, about this would be helpful, since the documentation is currently quite ambiguous in this regard.

alisitsyn, 2024-05-06:
@dmglogowskiOS,

Thank you for this issue. It is a very good question. Unfortunately, the current API do propagate the error code to indicate an exception but do not provide the way to determine the exact code of exception and the way of handling for this. I don't have nice solution for this but can provide the simple workarounds for this issue. One solution will not involve the changes in the library code.
I will provide update ASAP.

alisitsyn, 2024-05-10:
@dmglogowskiOS,

I propose the workaround to add [the user error handler function](https://github.com/espressif/esp-modbus/pull/62/commits/17de1dd9dd265313a604ae765ff431198f813a8d#diff-dd92830e4ffcf57adb5d49bd716a026fea482b21638a01919a9be86cb9b472dbR494) to catch the exception code. It also allows to check the data buffers on user side this can be useful for some gateways.

Please take a look and let me know if this would work for you.

dmglogowskiOS, 2024-05-13:
This may work quite well for most users imo, but I am unsure if it would be the best solution for my use case since I am working on a more object oriented (and C++ based) library based on esp-modbus and I am unsure how to implement this error handler in a manner that is compatible with my approach, although I do believe it should be possible.

alisitsyn, 2024-05-13:
> I am unsure if it would be the best solution for my use case since I am working on a more object oriented (and C++ based) library based on esp-modbus and I am unsure how to implement this error handler in a manner that is compatible with my approach, although I do believe it should be possible.

I understand your requirement. It is still possible to implement this with the callback but can agree this is like workaround but not the nice approach to do. The best approach would be to propagate the possible exception codes to the subset error codes returned from  `mbc_serial_master_send_request()` but the original idea was to hide Modbus specifics (mb exceptions are) from higher layer and I would not want to change it now. Other way could be to implement some additional API to get the exception code related to latest completed Modbus IO request. This would change the API interface but in compatible way. Also, the approach selected for this needs to be implemented the same way in v2.0 stack. I am still thinking about better way for this. You can also propose your view to this.

dmglogowskiOS, 2024-05-13:
I believe a possible way to handle this might be by adding either a function `mbc_serial_master_last_return()` which should return the last state (Either 0 or the exception code of the last exception) or adding an overload to `mbc_serial_master_send_request()`, either by making it an option configurable in the menuconfig or by adding a separate function (`mbc_serial_master_send_request_ret()` maybe?) where in addition to the request and the data, you also pass it a pointer to a return/metadata buffer. 

What could also work is by maybe refactoring the function to return this struct:

```c
typedef struct {
    const esp_err_t err;
    const uint8_t function_code;
    const uint8_t exception_code;
} slave_return_t
```

The caller can then read the error code from that structure and handle any exception accordingly, if the error code signifies an exception occured.

What may definitively be needed is an error code excplicitly alerting the program that an exception has occured, rather than `ESP_FAIL`...

alisitsyn, 2024-05-14:
> Obviously this second approach would definitely break the current API.

This way will not be implemented.

> I believe a possible way to handle this might be by adding either a function

Your approach is a bit different with what I mentioned [the function mbc_master_get_transaction_info](https://github.com/espressif/esp-modbus/pull/62/files#diff-dd92830e4ffcf57adb5d49bd716a026fea482b21638a01919a9be86cb9b472dbR395)

dmglogowskiOS, 2024-05-14:
That function would be almost *exactly* what I need and I could easily integrate this into my library

alisitsyn, 2024-05-14:
Ok, good. This will be updated accordingly and let it keep as basic approach. I would keep the callback approach as well, because it can be useful in some other user use cases.

alisitsyn, 2024-05-15:
This [old CPP wrapper](https://github.com/alisitsyn/modbus_support/blob/mb_support/modbus_add_cpp_wrapper) will probably help you.

dmglogowskiOS, 2024-05-15:
Thanks but see my answer in the associated Repo.

alisitsyn, 2024-06-07:
The issue has been fixed with commit 1eafc29df3d14ec379ea76f07337ae9816443e75

alisitsyn, 2024-06-07:
The component release is a bit postponed for now.

SputnikTheMighty, 2026-11-12:
Hi @alisitsyn 

Thanks for your message, it helped a lot and I have now registered custom handlers for several modbus functions.

However, I noticed when there is a modbus exception, the custom handler does not get called. It is skipped in this function https://github.com/espressif/esp-modbus/blob/main/modbus/mb_objects/mb_master.c#L104

I am interested in getting the returned exception code only for failed reads, so this isn't very helpful. 

Would it be possible to also call the custom handlers with modbus exceptions too?

alisitsyn, 2026-11-12:
@SputnikTheMighty ,

I got this. Sorry, this is an open source project and the behavior of the code changes often as per issues from other users. This handler behavior was changed recently and I missed this. Honestly, I would say this will not be fixed quickly. I have other project and many activities and issues that related to AI but I am not AI and need time to fix this. I will reopen this issue then to address this accordingly. I would advice you to make the changes you needed to track the exception code in your local component code for now. There is code in v1 which can be ported into mb object state machine to expose all information from transaction queue but it decide to be obsolete due to function handlers approach.

alisitsyn, 2026-11-12:
Issue reopened due to new aspects (changes in the code that block the related functionality).

SputnikTheMighty, 2026-11-13:
@alisitsyn

Thanks for quick response! No worries about a quick fix.

For now, I have the patch below (I'm working on v2.1.3) to allow handlers to attach to both exception responses and non-exception responses.

```
diff --git a/modbus/mb_objects/mb_master.c b/modbus/mb_objects/mb_master.c
index 4d13c23..f2258a0 100644
--- a/modbus/mb_objects/mb_master.c
+++ b/modbus/mb_objects/mb_master.c
@@ -108,21 +108,40 @@ static mb_exception_t mbm_check_invoke_handler(mb_base_t *inst, uint8_t func_cod
     if (!func_code || !buf) {
         return MB_EX_ILLEGAL_FUNCTION;
     }
+    
     if (func_code & MB_FUNC_ERROR) {
+        /* Handle exception responses
+        Exception respnses are 2 bytes: 
+         - the function code with error flag
+         - and the exception code
+        if the buffer ain't big enough, it's an invalid PDU,
+        so return MB_EX_ILLEGAL_DATA_VALUE */
         if (!len || (*len <= MB_PDU_DATA_OFF)) {
             ESP_LOGE(TAG,
                      "Invalid truncated exception PDU, length=%u",
                      len ? (unsigned)*len : 0U);
             return MB_EX_ILLEGAL_DATA_VALUE;
         }
-        exception = (mb_exception_t)buf[MB_PDU_DATA_OFF];
-        return exception;
     }
     SEMA_SECTION(mbm_obj->handler_descriptor.sema, MB_HANDLER_UNLOCK_TICKS) {
         mb_fn_handler_fp handler = NULL;
         mb_err_enum_t status = mb_get_handler(&mbm_obj->handler_descriptor, func_code, &handler);
+        
         if ((status == MB_ENOERR) && handler) {
+            /* For all non-exception responses, there is a default handler,
+            so we should always end up running this line */
             exception = handler(inst, buf, len);
+        
+        } else if (func_code & MB_FUNC_ERROR) {
+            /* There are no default handlers for exception responses, but users may
+            implement custom handlers. If they don't, we end up here */
+            ESP_LOGD(TAG, "No handler found for exception response, func_code=0x%02x", func_code);
+            exception = (mb_exception_t)buf[MB_PDU_DATA_OFF];
+        
+        } else {
+            /* This should almost never run, only if the user has deleted a default
+            handler without adding a new one. */
+            ESP_LOGW(TAG, "No handler found for function code=0x%02x", func_code);
         }
     }
     return exception;
```
and
```
diff --git a/modbus/mb_objects/functions/mbfunc_handling.c b/modbus/mb_objects/functions/mbfunc_handling.c
index 1b6c85b..1ae0ddf 100644
--- a/modbus/mb_objects/functions/mbfunc_handling.c
+++ b/modbus/mb_objects/functions/mbfunc_handling.c
@@ -13,8 +13,10 @@ static const char TAG[] __attribute__((unused)) = "MB_FUNC_HANDLING";
 mb_err_enum_t mb_set_handler(handler_descriptor_t *descriptor, uint8_t func_code, mb_fn_handler_fp handler)
 {
     MB_RETURN_ON_FALSE((descriptor && handler && descriptor->instance), MB_EINVAL, TAG, "invalid arguments.");
-    MB_RETURN_ON_FALSE(MB_IS_VALID_FUNC_CODE(func_code), MB_EINVAL, TAG,
-                       "invalid function code (0x%x)", (int)func_code);
+    
+    /* Commented (13/11/2026) as we want to register exception handlers as well */
+    /* MB_RETURN_ON_FALSE(MB_IS_VALID_FUNC_CODE(func_code), MB_EINVAL, TAG,
+                       "invalid function code (0x%x)", (int)func_code); */
 
     mb_command_entry_t *item_ptr = NULL;
     LIST_FOREACH(item_ptr, &descriptor->head, entries) {
@@ -45,8 +47,10 @@ mb_err_enum_t mb_set_handler(handler_descriptor_t *descriptor, uint8_t func_code
 mb_err_enum_t mb_get_handler(handler_descriptor_t *descriptor, uint8_t func_code, mb_fn_handler_fp *handler)
 {
     MB_RETURN_ON_FALSE((descriptor && handler && descriptor->instance), MB_EINVAL, TAG, "invalid arguments.");
-    MB_RETURN_ON_FALSE(MB_IS_VALID_FUNC_CODE(func_code), MB_EINVAL, TAG,
-                       "invalid function code (0x%x)", (int)func_code);
+    
+    /* Commented (13/11/2026) as we want to register exception handlers as well */
+    /* MB_RETURN_ON_FALSE(MB_IS_VALID_FUNC_CODE(func_code), MB_EINVAL, TAG,
+                       "invalid function code (0x%x)", (int)func_code); */
 
     mb_command_entry_t *item_ptr = NULL;
     LIST_FOREACH(item_ptr, &descriptor->head, entries) {
```