espressif/esp-modbus #65: Error when add esp-modbus-release-v2.0 into my project

TiDiChi, 2024-07-03:
Hi my friend,
      I am create a project with modbus RS485. I added esp-modbus-release-v2.0 component into "components" folder of project and then i used command "idf.py fullclean", but when i built my project, it displayed error: "fatal error: mbcontroller.h: No such file or directory
    4 | #include "mbcontroller.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
BUG: component_requirements.py: mbcontroller.h found in component esp-modbus-release-v2.0 which is already in the requirements list of ses-device".
    I don't understand this error because i add component esp-modbus-release-v2.0 to requirements list of ses-device component.

closed 2024-07-10 by TiDiChi

alisitsyn, 2024-07-19:
Hi @TiDiChi, 
Have you been able to solve the issue with esp-modbus v2.0?
In order to create the project based on v2:
1. Copy the example code from [here](https://github.com/espressif/esp-modbus/tree/release/v2.0/examples/serial/mb_serial_master) to your `root/mb_serial_master` project folder.
2. Copy the following folders and files into `proj_root/components/esp-modbus_v2` folder from [esp-modbus_v2 release](https://github.com/espressif/esp-modbus/tree/release/v2.0).
esp-modbus/modbus
esp-modbus/CMakeLists.txt
esp-modbus/component.mk
esp-modbus/idf_component.yml
esp-modbus/Kconfig
esp-modbus/LICENSE
esp-modbus/linker.lf
esp-modbus/pytest.ini
esp-modbus/README.md  
3. Copy the [common data library](https://github.com/espressif/esp-modbus/tree/release/v2.0/examples/mb_example_common) into root of the v2 folder (where the master project is located).
4. change the `mb_serial_master/main/idf_component.yml` as below:
``` yml
dependencies:
  idf: ">=4.1"
  mb_example_common:
    path: "../../mb_example_common"
```
5. Type the commands in the terminal: "idf.py fullclean;idf.py build"

You should be able to build the project.
Let me know if you have any questions.

alisitsyn, 2024-09-04:
Hi @TiDiChi,

The `esp-modbus v2.0.0-beta.1` component is released.
The updated version supports the multi-instance TCP Slave and Master with its own communication options. The pre-release esp-modbus component version [2.0.0-beta.1](https://components.espressif.com/components/espressif/esp-modbus/versions/2.0.0-beta.1) can be selected by `idf_component.yml` in main folder like below:

``` yml
dependencies:
  idf:
     version: ">=4.1.0"
  espressif/esp-modbus:
    version: "^2.0.0-beta"
  espressif/mdns:
    version: "^1.0.0"
    rules:
      - if: "idf_version >=5.0"
  mb_example_common:
    path: "../mb_example_common"
  protocol_examples_common:
    path: ${IDF_PATH}/examples/common_components/protocol_examples_common
```

alisitsyn, 2024-09-06:
@TiDiChi, 
Have you been able to solve your issue with the component?