espressif/esp-modbus #66: Has "mbfuncinput.c" file a bug?

NOM-99, 2024-07-18:
Tools Used:
 - IDE       -> VS-2019 16.11.33 / VisualGDB-5.6R9-4777 over Windows 11.
- Toolchain -> 8.4.0/8.1.0/r9
 - Framework -> ESP-IDF v4.4.3

Good afternoon,

Right now, I am using the "esp-modbus" component to work as a slave with a Modbus-RTU communication, 9600 baud rate, without parity and with two stop bits. I am triying to read 125 Holding registers and 125 Input registers.

When the Holding registers are read the communication is correct between the master and the slave; however, when the Input registers are read the communication returns "Modbus illegal data value" messages. Therefore, investigating the files that manage these communication functions I have found that in the Holding registers reading function the maximum number of registers that can be read are 125 and in the Input registers reading function the maximum number of registers that can be read are 124. 

Screenshots of "**mbfuncholding.c**" file of the "**eMBFuncReadHoldingRegister()**" function:

[image]
[image]

Screenshots of "**mbfuncinput.c**" file of the "**eMBFuncReadInputRegister()**" function:

[image]
[image]

I have read the documentation "Modbus application protocol specification" and the reading limit of the Holding and Input registers should be the same, 125.

[image]
[image]

I would like to know if this is really a bug of the file "mbfuncinput.c" or if this limit has become different for some particular reason.

Thank you very much.

NOM-99, 2024-07-19:
Thank you very much.

alisitsyn, 2024-07-29:
@NOM-99 ,

Thank you for your attention to this.

closed 2024-08-29 by alisitsyn