espressif/esp-modbus #69: mbc_master_get_parameter参数的意义

johnzhy, 2024-08-16:
不太明白传递这些参数的目的是什么，cid、name，type都包含在param_descriptor中，还要分开传递，而且一旦cid确定，实际上所有参数都是确定的，也不能传递错误，cid也必须对应相应的name
如果说想通过cid获取name、value、type，那应该允许传递空参数，比如我并不想知道name，type，只想简单的通过cid获取value

这么设计有什么特殊目的，或者不用这个方法，有更简单的使用方法

johnzhy, 2024-08-16:
而且既然都包装了这么复杂的接口，为什么不把多线程也考虑进去，比如另一个任务想要方便的获取读取到的传感器数据，或者另一个任务想要写入某个变量。

alisitsyn, 2024-08-29:
> 这么设计有什么特殊目的，或者不用这个方法，有更简单的使用方法

> 不太明白传递这些参数的目的是什么，cid、name，type都包含在param_descriptor中，还要分开传递，而且一旦cid确定，实际上所有参数都是确定的，也不能传递错误，cid也必须对应相应的name
如果说想通过cid获取name、value、type，那应该允许传递空参数，比如我并不想知道name，type，只想简单的通过cid获取value

This is intentionally designed. This additional checks (key) are performed in addition CID to check that the data dictionary is actual for current parameter being accessed. Just assume that the user starts communication and then changes the data dictionary in a unsafe way. This API interface is used up to this time to keep original interface unchanged esp-modbus v1.0.x (requirement). The API interface of [esp-modbus `v2.0.0-beta`](https://github.com/espressif/esp-modbus/blob/release/v2.0/modbus/mb_controller/serial/mbc_serial_master.c#L380) is changed and do not have additional check for the key.

alisitsyn, 2024-09-06:
@johnzhy, the solution proposed in v2.0.0-beta. Does this fix your issue?

johnzhy, 2024-09-06:
yes,i'll close this issue

closed 2024-09-06 by johnzhy

alisitsyn, 2024-09-06:
Thank you for update!