espressif/esp-modbus #70: 如何运行多个master？How can I run multiple masters?

zhuangqingshu, 2024-08-28:
假设我要在UART 1运行master_rtu 1,UART 2运行master_rtu 2，这能做到吗？
Suppose I want to run master_rtu 1 on UART 1 and master_rtu 2 on UART 2. Can this be done?

另一种情况是，UART 1运行master_rtu 1，再运行一个master_tcp，这能做到吗？
Another situation is that master_rtu 1 runs on UART 1 and then a master_tcp is run. Can this be done?

alisitsyn, 2024-08-29:
> Another situation is that master_rtu 1 runs on UART 1 and then a master_tcp is run. Can this be done?

closed 2024-08-30 by zhuangqingshu

alisitsyn, 2024-09-04:
@zhuangqingshu , 
The `esp-modbus v2.0.0-beta.1` component is released.
The updated version supports the multi-instance TCP Slave and Master with its own communication options. The pre-release esp-modbus component version [2.0.0-beta.1](https://components.espressif.com/components/espressif/esp-modbus/versions/2.0.0-beta.1) can be selected by `idf_component.yml` in main folder like below:

``` yml
dependencies:
  idf:
     version: ">=4.1.0"
  espressif/esp-modbus:
    version: "^2.0.0-beta"
  espressif/mdns:
    version: "^1.0.0"
    rules:
      - if: "idf_version >=5.0"
  mb_example_common:
    path: "../mb_example_common"
  protocol_examples_common:
    path: ${IDF_PATH}/examples/common_components/protocol_examples_common
```