espressif/esp-modbus #72: Data endianness problem

ForgetCSX, 2024-09-13:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

I encountered a format issue with the data endianness when using this repository.
[image]
[image]
When I read the holding register at address 0, the data returned to me is 85 1F 3F AB
Converting 1.34 through dotted hexadecimal converter the single precision float big endian format I get is 3F AB 85 1E
The float little endian format is 1E 85 AB 3F
This will cause my modbus master device to not recognize the received data

This is the data online converter I use
https://www.asciim.cn/hex/float.html

There is another question. I use the 06 command to write a single register. In the same way, I still operate the register at address 0.
[image]
As you can see, the writing should be successful, and the data returned to me is also 100, but when printing on the slave machine, there is still no change.
[image]

alisitsyn, 2024-09-16:
@ForgetCSX,

The frame logger shows the raw ADU of Modbus frame. As per [Modbus specification](https://www.modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf):
[image].
This means you should not expect the same order of the data as you described above because the Modbus standard standard does not prescribe how the extended data types mapping and it can be performed differently by device vendors and software. The esp-modbus Slave by default converts the 32-bit data from `big-endian` to the 'Little-endian byte swap' format understandable by compiler.
However, the esp-modbus supports [the data conversion functions](https://docs.espressif.com/projects/esp-modbus/en/latest/esp32/applications_and_references.html#modbus-endianness-conversion-api-reference) and you can convert the data to the format that is needed for your particular Slave application. The extended types support can be enabled by `CONFIG_FMB_EXT_TYPE_SUPPORT=y` in the kconfig menu. 
[image]

See the [esp-modbus extended datatypes](https://docs.espressif.com/projects/esp-modbus/en/latest/esp32/overview_messaging_and_mapping.html#mapping-of-complex-data-types), [Modbus Endianness Conversion API Reference](https://docs.espressif.com/projects/esp-modbus/en/latest/esp32/applications_and_references.html#modbus-endianness-conversion-api-reference), [example of conversion API](https://github.com/espressif/esp-modbus/blob/master/test/serial/mb_serial_slave/main/slave.c#L87). If you are using the esp-modbus master with your slaves, you can configure [the master to use appropriate format option](https://docs.espressif.com/projects/esp-modbus/en/latest/esp32/master_api_overview.html#id3) to work with your slave and the extended data conversion will be performed automatically by master in this case.

ForgetCSX, 2024-09-16:
@alisitsyn Thank you for your reply. I understand. Generally speaking, the problem is the data format. I didn’t know where to adjust it at first. I can adjust the byte order and it will be no problem. Now there is no doubt.

alisitsyn, 2024-09-30:
The issue is closed as explained.

closed 2024-09-30 by alisitsyn