espressif/esp-modbus #75: Add support for custom function codes.

klasrpersson, 2024-09-19:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

Add a way to write data to modbus slaves using arbitrary function codes. It could have an interface similar to eMBMasterReqWriteMultipleHoldingRegister but without the hardcoded function code.

### Use cases

I need to upgrade the firmware of a modbus device and it uses function codes 0x13 and 0x14 to download and upgrade firmware respectively. These function codes are not supported today and i cannot find a way to send in custom function codes.

### Additional context

I dont know if these function codes are established function codes for this purpose or if it is specific to the device im working with.

alisitsyn, 2024-09-20:
@klasrpersson,

Thanks you for the issue. Unfortunately, I can not add support of huge list of all Modbus commands into this stack.

The functions 0x13 and 0x14 included in the Modbus specification but are in the reserved list of commands (not standard). The support of custom commands and its handlers is not provided for now but this request is already added into feature list to implement. However, it is possible to do workaround in your application code to support your new commands without any changes in the stack code:
There is the function [mbc_master_send_request()](https://docs.espressif.com/projects/esp-modbus/en/latest/esp32/master_api_overview.html#master-communication), [the api description ](https://docs.espressif.com/projects/esp-modbus/en/latest/esp32/applications_and_references.html#_CPPv423mbc_master_send_requestP18mb_param_request_tPv) which allows to send custom request to any slave. However, it requires to have also the specific function handler. There is a way to handle response from slave to your custom command using the [custom user handler function] (https://docs.espressif.com/projects/esp-modbus/en/latest/esp32/master_api_overview.html#id5) in your application. This function is executed by stack right before return from [mbc_master_set_parameter()](https://docs.espressif.com/projects/esp-modbus/en/latest/esp32/applications_and_references.html#_CPPv424mbc_master_set_parameter8uint16_tPcP7uint8_tP7uint8_t) or [mbc_master_get_parameter()](https://docs.espressif.com/projects/esp-modbus/en/latest/esp32/applications_and_references.html#_CPPv424mbc_master_get_parameter8uint16_tPcP7uint8_tP7uint8_t) functions. This user handler can allow you to get the command send/receive buffer for each transaction. For your new functions the parameter (err_type = EV_ERROR_EXECUTE_FUNCTION) because the handler will not be found in the standard function table. So, it is possible to implement your own custom function handler for the functionality to support the functions 0x13, 0x14. Note: the custom handler should not blocked for a long time, so it is better to send the required buffers to the file handling task using the queues, then return from the handler function ASAP. It is responsibility of your application to handle the command properly.

I think this approach could work for you.

Please investigate the above and let me know if this can  work for you. 

Thanks.

alisitsyn, 2025-04-07:
@klasrpersson , the support for custom commands is merged with commit [cbe96056f2bcb44d4580355126b7255af332c3ba](https://github.com/espressif/esp-modbus/commit/cbe96056f2bcb44d4580355126b7255af332c3ba). This allows to add custom handler or override existing one for each Slave or Master object. If you have time, please take a look and verify this functionality. This will be available in next release.

The issue will be close as completed.

closed 2025-04-07 by alisitsyn