espressif/esp-modbus #79: Wrong frame format on serial RTU

mdjavedakhtar, 2024-10-20:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

Hello
i am trying to read data from a smart meter. the same setup is working with Arduino library(code attached)
but with esp component its not working also the bits are different in logic analyzer. can anyone suggest

**IDF code in app_main (uart setup is done before this)**
`
uint16_t val[4];
esp_err_t err;
uint16_t reg_type = 2;
uint16_t start_addr = 0x14;
uint16_t nRegs = 2;

	while(1) {
		err = modbus_get(start_addr, nRegs, reg_type, &val);
		if(err == ESP_OK)
			ESP_LOGI(TAG, "Val read: %d", val[1]);
		else
			ESP_LOGE(TAG, "Modbus read failed");

		vTaskDelay(1000 / portTICK_PERIOD_MS);
	}
`
**modbus_get Function**
`
esp_err_t modbus_get(uint16_t start_reg, uint16_t nRegs, uint16_t reg_type, void* dstBuf) {
	esp_err_t err;
	double slave_addr = 1;
	err = mbc_master_send_request(
		&(mb_param_request_t) {slave_addr, reg_type, start_reg, nRegs}, dstBuf);

	return err;
}
`

**Arduino IDE Code that works (modbus-esp8266 Library)**
`
mb.readHreg(1, 0x14, val, 2, cb); // Slave id is 1 and register address is 0x14
`
**logic analyzer screenshot from IDF code**
[image]

**logic analyzer screenshot from Arduino IDE code**
[image]

alisitsyn, 2024-10-21:
The function [mbc_master_send_request()](https://docs.espressif.com/projects/esp-modbus/en/latest/esp32/applications_and_references.html#_CPPv423mbc_master_send_requestP18mb_param_request_tPv) works if it is used properly.  

The example:
``` c
    // The slave uid, command to send, start register
    mb_param_request_t req = {.slave_addr = 1, .command = 0x03, .reg_start = 1, .reg_size = 2};

    uint8_t buf[4] = {0};

    err = mbc_master_send_request(&req, &info_buf[0]);
    if (err != ESP_OK) {
        ESP_LOGE("SEND_REQ", "Read slave info fail.");
    } else {
        ESP_LOGI("SEND_REQ", "Slave Data: %" PRIX32, *(uint32_t*)&buf[0]);
    }
```

mdjavedakhtar, 2024-10-21:
> The function [mbc_master_send_request()](https://docs.espressif.com/projects/esp-modbus/en/latest/esp32/applications_and_references.html#_CPPv423mbc_master_send_requestP18mb_param_request_tPv) works if it is used properly.  
> 
> The example:
> ``` c
>     // The slave uid, command to send, start register, register size
>     mb_param_request_t req = {.slave_addr = 1, .command = 0x03, .reg_start = 1, .reg_size = 2};
> 
>     uint8_t buf[4] = {0};
> 
>     err = mbc_master_send_request(&req, &info_buf[0]);
>     if (err != ESP_OK) {
>         ESP_LOGE("SEND_REQ", "Read slave info fail.");
>     } else {
>         ESP_LOGI("SEND_REQ", "Slave Data: %" PRIX32, *(uint32_t*)&buf[0]);
>     }
> ```

Hi
Thanks for the information
I just missed the command thing and it gave all kinds of strange issues there !!

closed 2024-10-21 by mdjavedakhtar