espressif/esp-modbus #81: Change the modbusTCP slave port number

iamugly, 2024-11-04:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

I would like to provide a way to modify the modbusTCP port number

### Use cases

Some scenarios may need to be modified

alisitsyn, 2024-11-05:
@iamugly,

The slave Modbus port is configurable through the [kconfig value](https://github.com/espressif/esp-modbus/blob/master/Kconfig#L9) in stack version v1.0.x. For the stack v2.0.0-beta, the TCP port is configurable independently in [the slave configuration structure](https://github.com/espressif/esp-modbus/blob/release/v2.0/examples/tcp/mb_tcp_slave/main/tcp_slave.c#L396) for each instance of master and slave.

alisitsyn, 2024-11-07:
@iamugly,

Does this solve the issue you described? Could you provide some feedback?

alisitsyn, 2024-12-02:
The issue is closed due to lack of feedback.

closed 2024-12-02 by alisitsyn