espressif/esp-modbus #82: How can I get the raw data sent by the master when using Modbus TCP as a client?

iotPeanut, 2024-11-19:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

How can I get the raw data sent by the master when using Modbus TCP as a client?

alisitsyn, 2024-11-19:
@iotPeanut,

Plese refer to [esp-modbus documentation, expose extra information](https://docs.espressif.com/projects/esp-modbus/en/latest/esp32/master_api_overview.html#expose-extra-information).

alisitsyn, 2024-12-05:
@iotPeanut,

Is the issue resolved on your side? I am going to close the issue. Feel free to reopen if you have update.

closed 2024-12-05 by alisitsyn