espressif/esp-modbus #84: Memory leakage between master_init() and mbc_master_destroy()

kirillstarodubov, 2024-11-21:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Memory allocated during master_init() shall be freed by mbc_master_destroy()

### Actual behavior (suspected bug)

in the line:
mbm_opts->mbm_event_group = xEventGroupCreate();
allocated memory each time master_init() is executed

### Steps to reproduce the behavior

emable HEAP_TRACING_STANDALONE in config
add heap tracing in the code
ESP_ERROR_CHECK( heap_trace_init_standalone(trace_record, NUM_RECORDS) );
ESP_ERROR_CHECK( heap_trace_start(HEAP_TRACE_LEAKS) );
launch example and call only two functions master_init() and mbc_master_destroy() between heap_trace_start() and heap_trace_stop()
ESP_ERROR_CHECK( heap_trace_stop() );
heap_trace_dump();

4kb will "leak" in  mbm_opts->mbm_event_group = xEventGroupCreate();

### Project release version

1.0.16

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

esp32

### Shell

Bash

alisitsyn, 2024-11-22:
@kirillstarodubov,

thank you for the reported issue.
The event group is correctly freed [here](https://github.com/espressif/esp-modbus/blob/master/freemodbus/serial_master/modbus_controller/mbc_serial_master.c#L129). I guess the leaked 4k is the task stack. I think the information from heap trace is a bit incorrect. I will check this a bit later.

kirillstarodubov, 2024-11-22:
It is freed, but not freed in reality. If you loop several times the
init/destroy sycle you will see that memory is not properly freed despite
explicit free() call

alisitsyn, 2024-11-22:
Do you see some error messages in the log with debug verbosity?
> If you loop several times the init/destroy sycle you will see that memory is not properly freed despite explicit free() call

If so, it is an issue against destroy sequence and then you should see the error messages that the stack was not able to disable lower layers for example. I think the original root reason is hidden here. It was tested previously and worked just fine with no any leak. However, I got your issue and will check when possible.

kirillstarodubov, 2024-11-22:
no errors, free calls are executed

alisitsyn, 2024-11-22:
it is really strange.

alisitsyn, 2024-11-25:
Unfortunately, I was not able to reproduce your issue. See log above:
```
#include "esp_heap_trace.h"
#define NUM_RECORDS 200
static heap_trace_record_t trace_record[NUM_RECORDS]; // This buffer must be in internal RAM

void app_main(void)
{
    ESP_ERROR_CHECK( heap_trace_init_standalone(trace_record, NUM_RECORDS) );
    // Initialization of device peripheral and objects
    ESP_ERROR_CHECK( heap_trace_start(HEAP_TRACE_LEAKS) );
    for (int i=0; i<3; i++) {
        ESP_LOGW("TEST_LEAKS", "Test loop #%d start.", i);
        ESP_ERROR_CHECK(master_init());
        vTaskDelay(10);
        ESP_ERROR_CHECK(mbc_master_destroy());
        vTaskDelay(10);
        ESP_LOGW("TEST_LEAKS", "Test loop #%d stop.", i);
    }
    ESP_ERROR_CHECK( heap_trace_stop() );
    heap_trace_dump();

    //master_operation_func(NULL);
}
```
``` log
I (273) heap_init: Initializing. RAM available for dynamic allocation:
I (280) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (286) heap_init: At 3FFB59E8 len 0002A618 (169 KiB): DRAM
I (292) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (298) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (305) heap_init: At 4009033C len 0000FCC4 (63 KiB): IRAM
I (313) spi_flash: detected chip: generic
I (316) spi_flash: flash io: dio
W (319) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (333) main_task: Started on CPU0
I (343) main_task: Calling app_main()
W (343) TEST_LEAKS: Test loop #0 start.
I (343) uart: queue free spaces: 20
I (393) MASTER_TEST: Modbus master stack initialized...
W (593) TEST_LEAKS: Test loop #0 stop.
W (593) TEST_LEAKS: Test loop #1 start.
I (593) uart: queue free spaces: 20
I (643) MASTER_TEST: Modbus master stack initialized...
W (843) TEST_LEAKS: Test loop #1 stop.
W (843) TEST_LEAKS: Test loop #2 start.
I (843) uart: queue free spaces: 20
I (893) MASTER_TEST: Modbus master stack initialized...
W (1093) TEST_LEAKS: Test loop #2 stop.
====== Heap Trace: 3 records (200 capacity) ======
    84 bytes (@ 0x3ffb8b90, Internal) allocated CPU 0 ccount 0x0350029c caller 0x400820cb:0x40089208:0x40088369:0x400885d7
--- 0x400820cb: heap_caps_malloc at /home/esp-man/esp32/esp-idf/components/heap/heap_caps.c:84
0x40089208: pvPortMalloc at /home/esp-man/esp32/esp-idf/components/freertos/heap_idf.c:58
0x40088369: xQueueGenericCreate at /home/esp-man/esp32/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:545
0x400885d7: xQueueCreateMutex at /home/esp-man/esp32/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:669

    84 bytes (@ 0x3ffb9d50, Internal) allocated CPU 0 ccount 0x03521e58 caller 0x400820cb:0x40089208:0x40088369:0x400885d7
--- 0x400820cb: heap_caps_malloc at /home/esp-man/esp32/esp-idf/components/heap/heap_caps.c:84
0x40089208: pvPortMalloc at /home/esp-man/esp32/esp-idf/components/freertos/heap_idf.c:58
0x40088369: xQueueGenericCreate at /home/esp-man/esp32/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:545
0x400885d7: xQueueCreateMutex at /home/esp-man/esp32/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:669

    12 bytes (@ 0x3ffba698, Internal) allocated CPU 0 ccount 0x035a8a6c caller 0x400820cb:0x400d35b5:0x400d39a2:0x400d3bac
--- 0x400820cb: heap_caps_malloc at /home/esp-man/esp32/esp-idf/components/heap/heap_caps.c:84
0x400d35b5: get_desc_for_int at /home/esp-man/esp32/esp-idf/components/esp_hw_support/intr_alloc.c:156
0x400d39a2: esp_intr_alloc_intrstatus at /home/esp-man/esp32/esp-idf/components/esp_hw_support/intr_alloc.c:555
0x400d3bac: esp_intr_alloc at /home/esp-man/esp32/esp-idf/components/esp_hw_support/intr_alloc.c:671

====== Heap Trace Summary ======
Mode: Heap Trace Leaks
180 bytes 'leaked' in trace (3 allocations)
records: 3 (200 capacity, 28 high water mark)
total allocations: 78
total frees: 84
================================
I (1143) main_task: Returned from app_main()
```
Could you create reproducible example for this?

alisitsyn, 2024-12-02:
@kirillstarodubov, do you have any update related to this?

kirillstarodubov, 2024-12-02:
Sorry, Im a bit busy with other things. Ill add more info in coming days

kirillstarodubov, 2024-12-05:
I've repeated the test and it looks like i can not reproduce an issue
anymore, i suspect that my old config has smth to do with that...
neither my almost full code nor "only" modbus creates a leakage anymore.

W (17468) mb_bess: MODBUS MASTER stack initialized...

====== Heap Trace: 0 records (100 capacity) ======

====== Heap Trace Summary ======

Mode: Heap Trace Leaks

0 bytes 'leaked' in trace (0 allocations)

records: 0 (100 capacity, 28 high water mark)

total allocations: 25

total frees: 28
================================

Please close an issue

alisitsyn, 2024-12-05:
Thanks for report. The issue is closed.

closed 2024-12-05 by alisitsyn