espressif/esp-modbus #85: Modbus Serial Setup Error

praveens-git, 2024-11-24:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

I having trouble initialising modbus rtu in my code.

My Initialisation Code:

void rtu_master_init() 
{

    void *master_handler = NULL;

    esp_err_t err = mbc_master_init(MB_PORT_SERIAL_MASTER, &master_handler);
    if (master_handler == NULL)
    {
        ESP_LOGE(TAG, "mb controller initialization fail.");
        ESP_LOGE(TAG, "mb controller initialization fail, returns(0x%x).", (uint)err);
        while (1)
        {
        }
    }

    const mb_communication_info_t comm = {
        .mode = MB_MODE_RTU,
        .baudrate = 9600,
        .parity = UART_PARITY_DISABLE,
        .port = MB_PORT_NUM,
    };

    err = mbc_master_setup((void *)&comm);

    if (err != ESP_OK)
    {
        ESP_LOGE(TAG, "mb controller setup fail, returns(0x%x).", (uint)err);
        while (1)
        {
        }
    }

    // Set UART pin numbers
    err = uart_set_pin(MB_PORT_NUM, RS485_TX, RS485_RX, RS485_DE, UART_PIN_NO_CHANGE);
    if (err != ESP_OK)
    {
        ESP_LOGE(TAG, "mb serial set pin failure, uart_set_pin() returned (0x%x).", (uint)err);
        err = mbc_master_start();
        while (1)
        {
        }
    }

    if (err != ESP_OK)
    {
        ESP_LOGE(TAG, "mb controller start fail, returns(0x%x).", (uint)err);
        while (1)
        {
        }
    }

    err = uart_set_mode(MB_PORT_NUM, UART_MODE_RS485_HALF_DUPLEX);
    if (err != ESP_OK)
    {
        ESP_LOGE(TAG, "mb serial set mode failure, uart_set_mode() returned (0x%x).", (uint)err);
        vTaskDelay(5 / portTICK_PERIOD_MS);
        while (1)
        {
        }
    }

    err = mbc_master_set_descriptor(&rtu_device_parameters[0], rtu_num_of_params);
    if (err != ESP_OK)
    {
        ESP_LOGE(TAG, "mb controller set descriptor fail, returns(0x%x).", (uint)err);
        ESP_LOGI(TAG, "Modbus master stack initialized...");
        while (1)
        {
        }
    }
}

Error:

`E (1106) MB_CONTROLLER_MASTER: mbc_serial_master_setup(79): mb wrong port to set = (2457600).`
`E (1116) MB_CONTROLLER_MASTER: mbc_master_setup(159): Master setup failure, error=(0x102) (ESP_ERR_INVALID_ARG).`
`E (1126) RTU Master: master_init(153): mb controller setup fail, returns(0x102).`

May I know why this happens, I am just copying the example code and modified it a simple function for me to use

kabirwbt, 2024-11-25:
@praveens-git you have to define MB_PORT_NUM in your code! If you are using esp-idf then use the sdkconfig to define it or us a #define in your code to define it to 1/2/3 based on your hardware.

praveens-git, 2024-11-25:
> which UART you are using

I have defined that.

alisitsyn, 2024-11-25:
@praveens-git,

> May I know why this happens, I am just copying the example code and modified it a simple function for me to use

The initialization of Modbus looks ok in the code above. The error above shows that your serial port number s set to incorrect value. The `2457600` is seems the value of the cristal oscillator in some of your code. Please check your sdkconfig file for correctness.

alisitsyn, 2024-12-05:
@praveens-git,

Any update on this issue?

praveens-git, 2024-12-06:
Well, as a debug I printed the pointed memory, and it seems like misalignment so as of now, I packed the structure and looks it is working now

alisitsyn, 2024-12-06:
@praveens-git ,
Thank you for feedback. The issue is closed then.

closed 2024-12-06 by alisitsyn

praveens-git, 2024-12-06:
But why this happens in the library.

alisitsyn, 2024-12-06:
> Well, as a debug I printed the pointed memory, and it seems like misalignment so as of now, I

How I can answer to your question if you do not describe what goes wrong with the library, which structures have misalignment in your code. The common  register related structures have the `#pragma pack(push, 1)` to be aligned to 1 byte scope it is required, for others not required. As I described above the error message `E (1106) MB_CONTROLLER_MASTER: mbc_serial_master_setup(79): mb wrong port to set = (2457600).` clearly says it is something wrong with your port setting. Please set it explicitly to the port number you use and call the API this should work just fine. 
Please provide the logs or results of investigation that show the structures and confirm their misalignment or create the simple example to reproduce the issue. Thanks you.