espressif/esp-modbus #88: RTU ModBus unexpected event triggered 0x02 / Timeout before any socket was ready!

sorinvn, 2025-01-12:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I am using ESP32-C3 Modbus RTU with modbus version 2.0.1 for communication with Schneider iem3150 energy meter.
Every 3 seconds I get an " unexpected event triggered", nevertheless both reading and writing registers work fine. 

### Actual behavior (suspected bug)

Please see below error log (debug enabled). Timeout messages start to occur immediately after modbus initialization. Before any write or read operation, the error message is "unexpected event triggered 0x02", after performing a write operation the error message changes to "unexpected event triggered 0x40".

This error occurs aftert getting "transport_base: poll_read: select - Timeout before any socket was ready!" every second. From what see, this message is triggered by the TCP communication stack (?!) and is present both with Modbus TCP enabled or disabled in sdkconfig (only RTU is used). 

Additionally (independent issue), the first write message also shows the behavior described in IDFGH-10848, nevertheless after retry the write operation is done correctly.

The issue is present with modbus v2.0 and 2.0.1. It was not present with v1.17.
This issue is not critical since both read and write are ok, so it looks like a nuisance message.

### Error logs or terminal output

```shell
--- esp-idf-monitor 1.5.0 on /dev/cu.usbserial-1110 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xf (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1574
load:0x403cc710,len:0xc30
load:0x403ce710,len:0x2f58
entry 0x403cc71a
I (30) boot: ESP-IDF v5.4-dirty 2nd stage bootloader
I (30) boot: compile time Jan 10 2025 18:21:17
I (30) boot: chip revision: v0.4
I (31) boot: efuse block revision: v1.3
I (34) boot.esp32c3: SPI Speed      : 80MHz
I (38) boot.esp32c3: SPI Mode       : DIO
I (42) boot.esp32c3: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00180000
I (79) boot: End of partition table
I (82) esp_image: segment 0: paddr=00010020 vaddr=3c0d0020 size=29eb0h (171696) map
I (117) esp_image: segment 1: paddr=00039ed8 vaddr=3fc94e00 size=02f48h ( 12104) load
I (119) esp_image: segment 2: paddr=0003ce28 vaddr=40380000 size=031f0h ( 12784) load
I (123) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=c528ch (807564) map
I (257) esp_image: segment 4: paddr=001052b4 vaddr=403831f0 size=11b18h ( 72472) load
I (270) esp_image: segment 5: paddr=00116dd4 vaddr=50000200 size=0001ch (    28) load
I (277) boot: Loaded app from partition at offset 0x10000
I (277) boot: Disabling RNG early entropy source...
I (288) cpu_start: Unicore app
I (296) cpu_start: Pro cpu start user code
I (296) cpu_start: cpu freq: 160000000 Hz
I (296) app_init: Application information:
I (296) app_init: Project name:     iem3150
I (300) app_init: App version:      1
I (304) app_init: Compile time:     Jan 10 2025 18:21:23
I (309) app_init: ELF file SHA256:  8e3b0563b...
[... 267 lines cut ...]
}

### Project release version

v2.0.1

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

MacOS

### Operating system version

macos 15.2 

### Shell

ZSH

### Additional context

IDF 5.3.2 or 5.4

alisitsyn, 2025-01-13:
Hi @sorinvn,

Thank you for this report.

> This error occurs aftert getting "transport_base: poll_read: select - Timeout before any socket was ready!" every second. From what see, this message is triggered by the TCP communication stack (?!) and is present both with Modbus TCP enabled or disabled in sdkconfig (only RTU is used). 

The message `transport_base: poll_read: select - Timeout before any socket was ready!` in your log is not related to the Modbus TCP stack but comes from `tcp_transport` component used in your other application code. Please let me know what component is used in your application (is it MQTT client)? According to the log I can guess that this component has a task on the same core with the priority higher than the Modbus task and it prevents Modbus task to trigger for longer than 1 second. This causes the Modbus events to expire. You should modify your TCP task to allow the Modbus task to trigger. Please modify the tasks sequence or configure the priorities of the active objects in your project properly to allow Modbus task to trigger. The kconfig value `CONFIG_FMB_PORT_TASK_PRIO` should be configured according to priorities of your other tasks.

Please provide more information on what kind of additional components are used and priorities of the tasks in your application. Also, please include extended Modbus log and sdkconfig file.

> Additionally (independent issue), the first write message also shows the behavior described in IDFGH-10848, nevertheless after retry the write operation is done correctly.

The issue is present with modbus v2.0 and 2.0.1. It was not present with v1.17.
This issue is not critical since both read and write are ok, so it looks like a nuisance message.

The v1.0.17 contains the commit which fixes the `mbc_master_send_request()` issue. Unfortunately, I could not backport all the fixes for v2.0.1 and will update it as soon as possible.

sorinvn, 2025-01-13:
Hello Alex, thanks for your prompt reply.
I'm using indeed mqtt and sntp.
Attached are the full log and sdkconfig file.
[sdkconfig.txt](https://github.com/user-attachments/files/18395780/sdkconfig.txt)
[log.txt](https://github.com/user-attachments/files/18395781/log.txt)

noted your comment re mbc_master_send_request.
One additional clarification, when I said these were not present in 1.17, I ment the unexpected event triggered was not present.

alisitsyn, 2025-01-13:
@sorinvn ,

> 'm using indeed mqtt and sntp.
Attached are the full log and sdkconfig file.

Thank you for the feedback.

> One additional clarification, when I said these were not present in 1.17, I ment the unexpected event triggered was not present.

Thanks for the clarification I will take a look to it as soon as I can. For now please try to follow the above recommendations to optimize your system to not block the Modbus for long time and let me know if you have any updates.

Thanks.

Fienberber, 2025-02-04:
Hello, 

I have the same exact issue using the ESP-IDF v5.2.2 

and Project release version v2.0.1

alisitsyn, 2025-02-28:
> Hello Alex, thanks for your prompt reply. I'm using indeed mqtt and sntp. Attached are the full log and sdkconfig file. [...]

The esp-modbus v1.0.17 uses infinite waiting time for events but v2 uses the finite time. However, again this is normal situation and messages are left for debugging purpose. The verbosity is changes to debug. This will be merged in v2.0.2.

alisitsyn, 2025-03-03:
The fix was merged with commit [895e9279]. See the [ESP_Modbus component release v2.0.2](https://github.com/espressif/esp-modbus/releases/tag/v2.0.2)

closed 2025-03-03 by alisitsyn