espressif/esp-modbus #89: Build error in component

Kontrolojik, 2025-01-16:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I expect it to compile

### Actual behavior (suspected bug)

Instead, simple project with version 2.0.1 dose not compile. Because file path for component cache is too long for windows. 
FileNotFoundError: [WinError 206] Dosya ad² veya uzant²s² ok uzun:
  'C:\\Users\\YilmazK\\AppData\\Local\\Espressif\\ComponentManager\\Cache\\service_d92d8f1e\\espressif__esp-modbus_2.0.1_1c3be72a\\examples\\pytest_embedded_log\\2024-12-20_09-38-42\\test_modbus_tcp_host_to_slave_communication[1-\\builds\\idf\\esp-modbus\\examples\\tcp'

### Error logs or terminal output

```shell
-- Found Git: D:/IDF/v5.4/tools/.espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/IDF/v5.4/tools/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/IDF/v5.4/tools/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: D:/IDF/v5.4/tools/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
NOTICE: Manifest files have changed, solving dependencies.
.......NOTICE: Updating lock file at D:\IDF\v5.4\projects\relexbox_v9\dependencies.lock
NOTICE: Processing 3 dependencies:
NOTICE: [1/3] espressif/esp-modbus (2.0.1)
CMake Error at D:/IDF/v5.4/esp-idf/tools/cmake/build.cmake:610 (message):
  Traceback (most recent call last):

    File "<frozen runpy>", line 198, in _run_module_as_main
    File "<frozen runpy>", line 88, in _run_code
    File "d:\IDF\v5.4\tools\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\idf_component_manager\prepare_components\__main__.py", line 6, in <module>
      main()
    File "d:\IDF\v5.4\tools\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 130, in main
      args.func(args)
    File "d:\IDF\v5.4\tools\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 28, in prepare_dep_dirs
      ).prepare_dep_dirs(
        ^^^^^^^^^^^^^^^^^
    File "d:\IDF\v5.4\tools\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\idf_component_manager\core.py", line 99, in wrapper
      return func(self, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "d:\IDF\v5.4\tools\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\idf_component_manager\core.py", line 717, in prepare_dep_dirs
      downloaded_components = download_project_dependencies(
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "d:\IDF\v5.4\tools\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\idf_component_manager\dependencies.py", line 404, in download_project_dependencies
      download_path = fetcher.download()
                      ^^^^^^^^^^^^^^^^^^
    File "d:\IDF\v5.4\tools\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\idf_component_tools\sources\fetcher.py", line 70, in download
      return self.source.download(self.component, self.managed_path)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "d:\IDF\v5.4\tools\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\idf_component_tools\sources\web_service.py", line 266, in download
      unpack_archive(file_path, self.component_cache_path(component))
    File "d:\IDF\v5.4\tools\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\idf_component_tools\archive_tools.py", line 98, in unpack_archive
      handler(file, destination_directory)
    File "d:\IDF\v5.4\tools\.espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\idf_component_tools\archive_tools.py", line 90, in unpack_zip
      archive.extract(item, destination_directory)
    File "zipfile.py", line 1660, in extract
    File "zipfile.py", line 1723, in _extract_member
    File "<frozen os>", line 215, in makedirs
    File "<frozen os>", line 225, in makedirs

  FileNotFoundError: [WinError 206] Dosya ad² veya uzant²s² ok uzun:
  'C:\\Users\\YilmazK\\AppData\\Local\\Espressif\\ComponentManager\\Cache\\service_d92d8f1e\\espressif__esp-modbus_2.0.1_1c3be72a\\examples\\pytest_embedded_log\\2024-12-20_09-38-42\\test_modbus_tcp_host_to_slave_communication[1-\\builds\\idf\\esp-modbus\\examples\\tcp'

Call Stack (most recent call first):
  D:/IDF/v5.4/esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!

 *  The terminal process "d:\IDF\v5.4\tools\.espressif\tools\cmake\3.30.2\bin\cmake.exe '-G', 'Ninja', '-DPYTHON_DEPS_CHECKED=1', '-DESP_PLATFORM=1', '-B', 'd:\IDF\v5.4\projects\relexbox_v9\build', '-S', 'd:\IDF\v5.4\projects\relexbox_v9', '-DSDKCONFIG=d:\IDF\v5.4\projects\relexbox_v9\sdkconfig', '-DCCACHE_ENABLE=1'" terminated with exit code: 1.
```

### Steps to reproduce the behavior

add dependecy to esp_modbus component. then make a simple project. 
My idf_component.yml is like this;
dependencies:
  joltwallet/littlefs: ^1.14.8
  espressif/esp-modbus: ^2.0.1

### Project release version

2.0.1

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11 pro

### Shell

other (details in Additional context)

### Additional context

I see this both in vscode terminal and ESP-IDF terminals as well.

alisitsyn, 2025-01-17:
Hi @Kontrolojik,

Thank you for this issue report. Please add your complete *.yml file here (just for extended information). This issue happens because during release stage the test related files are not excluded correctly from the component archive.This will be fixed in new version of the component. For now I think you can download end extract component files into components/esp-modbus folder in the root of your project and then remove the `pytest_embedded_log` folders manually.

Kontrolojik, 2025-01-17:
Hi Alisitsyn,

I cannot attach the yml file here, it says its not allowed but below is the whole content of the file;
## IDF Component Manager Manifest File
dependencies:
  joltwallet/littlefs: "^1.14.8"
  espressif/esp-modbus: ^2.0.1

I also realized that during build process cache was dealing with build files in test folders which should be git-ignored.

Can we expect a quick release like 2.0.2 with this fix instead of witing for the new release which might take time?

Regards

alisitsyn, 2025-01-17:
@Kontrolojik,

Thanks for update. 
> This can be happened only under Windows and all these files have to be ignored by filters in component `yml` file. The rules did not work due to issue.

I can do the update with the version of `2.0.1~1`, I will try to do this (need to combine all minors together) but still propose you to build your project with manually updated component for now.

colafati, 2025-01-19:
Same issue here, with both 2.0.0 an 2.0.1, it works with the 1.0.17. 
Windows, using ESP-IDF and Visual Studio Code.

alisitsyn, 2025-01-27:
Hi @colafati , @Kontrolojik,

The fix has been fixed [v2.0.1~1](https://components.espressif.com/components/espressif/esp-modbus/versions/2.0.1~1).
PTAL.

colafati, 2025-01-29:
I tried it with 2.0.1~1 and updated the domimartinglogi/dynamic_modbus_master; it's working.  Thanks @alisitsyn

alisitsyn, 2025-01-30:
@colafati, thank you for report. The issue will be closed then.

closed 2025-01-30 by alisitsyn