espressif/esp-modbus #91: including mbcontroller.h from a cpp file breaks compilation

attilabody, 2025-01-31:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Including mbcontroller.h from a cpp file should not break compilation

### Actual behavior (suspected bug)

Including mbcontroller.h from a cpp file breaks compilation

### Error logs or terminal output

```shell
In file included from /home/abody/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/atomic_base.h:36,
                 from /home/abody/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/atomic:50,
                 from /home/abody/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/stdatomic.h:36,
                 from /mnt/userdata_ssd/abody/work/heattransformers/RemoteMockHeatpump/managed_components/espressif__esp-modbus/modbus/mb_objects/include/mb_port_types.h:8,
                 from /mnt/userdata_ssd/abody/work/heattransformers/RemoteMockHeatpump/managed_components/espressif__esp-modbus/modbus/mb_ports/common/port_common.h:19,
                 from /mnt/userdata_ssd/abody/work/heattransformers/RemoteMockHeatpump/managed_components/espressif__esp-modbus/modbus/mb_controller/common/include/esp_modbus_common.h:20,
                 from /mnt/userdata_ssd/abody/work/heattransformers/RemoteMockHeatpump/managed_components/espressif__esp-modbus/modbus/mb_controller/common/include/esp_modbus_master.h:12,
                 from /mnt/userdata_ssd/abody/work/heattransformers/RemoteMockHeatpump/managed_components/espressif__esp-modbus/modbus/mb_controller/common/include/mbcontroller.h:17,
                 from /mnt/userdata_ssd/abody/work/heattransformers/RemoteMockHeatpump/main/remote_mock_heatpump_main.cpp:18:
/home/abody/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/new:195:3: error: template with C linkage
  195 |   template<typename _Tp>
      |   ^~~~~~~~
...
(hundreds of lines of errors follow)
```

### Steps to reproduce the behavior

1. activate esp-idf (tested with 5.3.2 and 5.4)
2. create a new project from the Hello World example
3. add the esp-modbus as a dependency
4. rename the main c file to cpp
5. adjust Cmake accordingly
6. add  "mbcontroller.h" to the includes
7. idf. py build

optional:
8. rename the main cpp file back to c
9. the build succeeds

### Project release version

^2.0.1~1

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Arch

### Shell

Bash

fink-at-trmc-dk, 2025-02-02:
Excactly the same problem. Cannot compile as C++...

alisitsyn, 2025-02-05:
The preliminary solution is provided in the PR above. Please check. Thanks.

fink-at-trmc-dk, 2025-02-09:
IT seems to compile on my setup now. Now I just need to migrate from 1.x to 2.x. But the compilation åroblems from before seems to be gone....

//Fink

alisitsyn, 2025-02-28:
The fix is merged to main branch with commit  931f554d.

closed 2025-03-03 by alisitsyn