espressif/esp-modbus #93: RTU on and ASCII off

diplfranzhoepfinger, 2025-02-09:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

should work if RTU on and ASCII off in sdk config. 

### Actual behavior (suspected bug)

```
CONFIG_FMB_COMM_MODE_RTU_EN=y
# CONFIG_FMB_COMM_MODE_ASCII_EN is not set
```

leads at Compile to: 

```
/managed_components/espressif__esp-modbus/modbus/mb_objects/mb_slave.c:148:11: error: implicit declaration of function 'mbs_ascii_transp_create'; did you mean 'mbs_rtu_transp_create'? [-Wimplicit-function-declaration]
```

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

just try. 
like above. 

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu

### Shell

Bash

alisitsyn, 2025-02-11:
@diplfranzhoepfinger, Thank you for reporting. The issue is known and the fix is [here](https://github.com/espressif/esp-modbus/pull/92/commits/5c0d66a8c0eeacbcb04ec4e26d15eac969220fa7). I am currently busy but going to merge the fix as soon as possible.

alisitsyn, 2025-02-27:
The fix has been  merged with the commit 9b24aa41.

closed 2025-02-27 by alisitsyn