espressif/esp-modbus #95: esp-clang支持?

DreamSilverFox, 2025-02-14:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

我使用ESP32 & "Espressif\tools\esp-clang\esp-18.1.2_20240912\esp-clang\bin" & **C++**去构建，但是失败了，看上去问题基本上是GCC和Clang的兼容性问题？
请问，有支持esp-clang的计划吗？

alisitsyn, 2025-02-20:
@DreamSilverFox, 

Thank you for the issue.

> 我使用ESP32 & "Espressif\tools\esp-clang\esp-18.1.2_20240912\esp-clang\bin" & **C++**去构建，但是失败了，看上去问题基本上是GCC和Clang的兼容性问题？
请问，有支持esp-clang的计划吗?

Yes, it is planned to add support of the `esp-clang` into `esp-modbus` project pipeline. Unfortunately, this was postponed for a while. As I remember the `esp-clang` was working just fine on my side. Please add more information (logs) to clarify your issue. I can ask the tools team to help with this issue if necessary. 

Thanks.

DreamSilverFox, 2025-02-21:
> [@DreamSilverFox](https://github.com/DreamSilverFox),
>
> Thank you for the issue.
>
>
> Yes, it is planned to add support of the `esp-clang` into `esp-modbus` project pipeline. Unfortunately, this was [...]

1. I find some problem from C++ , "IDFGH-14441" fix it.
2. **transp_obj** used before define, MB_RETURN_ON_FALSE, MB_GOTO_ON_FALSE
3. **MB_STR_CAT** *error: '}' and ')' tokens terminating statement expression appear in different macro expansion contexts [-Werror,-Wcompound-token-split-by-macro]*
4. **MB_OBJ_PARENT** *also '}' and ')' tokens terminating statement expression appear in different macro expansion contexts [-Werror,-Wcompound-token-split-by-macro]*

alisitsyn, 2025-02-21:
Thank you for feedback. I will take a look to it ASAP.

alisitsyn, 2025-07-29:
@DreamSilverFox,

The fix for clang was merged with commit 079b51e23b09f3e2254000336f310c9b4eb88c4a. The update has a tool script: tools/static_analyzer_check.sh that can be used to check gcc static analyzer or esp-clang. The CI jobs are also added to check the static analyzer warnings. The component will be released later. Please take a look to fix and let me know if you ave any issues.

Thank you for patience.

closed 2025-07-29 by alisitsyn