espressif/esp-modbus #102: Add "scalar" float to mb_parameter_opt_t struct

OllieKrause, 2025-03-07:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

Many of the Modbus registers I am reading need to be multiplied by a specified scaling factor to determine their true value. For example, 0x0050 holds an input voltage but its raw value is in decivolts and thus must be multiplied by a 0.1 scaling factor to convert back to volts (see attached image). It seems the most logical place to include this scaling factor would be as a new float within the mb_parameter_opt_t struct (this would represent opt4). 

[image]

### Use cases

Often values like voltage (eg. 12.56V) are stored as unsigned integers multiplied by some scaling factor specified in the datasheet. The useful value is obtained by multiplying the raw value at the register address by this scaling factor. See the attached datasheet for a meanwell NTN-5K inverter. 

[Mean Well NTN-5K-E Manual.pdf](https://github.com/user-attachments/files/19121452/Mean.Well.NTN-5K-E.Manual.pdf)

### Alternatives

I have considered solving this problem by defining a new struct `extended_mb_parameter_descriptor_t` that appends the scaling factor. 

`typedef struct {
mb_parameter_descriptor_t base; 
float scale_factor;    
} extended_mb_parameter_descriptor_t; `

This isn't exactly a clean solution though. 

### Additional context

I'm fairly new to modbus so please forgive me if there is a easier solution to this issue

alisitsyn, 2025-03-10:
Hello @OllieKrause, 

Thank you for your proposal. 

The option fields already contain the the int values that can be used as a divider for your scalar parameter in your application. Why do you think the esp-modbus needs additional field of float type?

OllieKrause, 2025-03-10:
Hi @alisitsyn! Thanks so much for your response! Sometimes the scalar is greater than one and other times it is less than one so a float with a decimal point is necessary. Alternatively, some other binary field is necessary to specify whether the raw value should be multiplied or divided by the integer scaling factor.

OllieKrause, 2025-03-11:
Hi @alisitsyn! Thanks again for your response. In that case, I think I'll use the first digit of the scalar to indicate a multiply or divide operation and the remaining digits to specify the scaling factor. I totally understand your need to maintain consistent formatting across versions. Thanks again for your help.

alisitsyn, 2025-03-12:
Hi @OllieKrause , 

> I think I'll use the first digit of the scalar to indicate a multiply or divide operation and the remaining digits to specify the scaling factor.

This can work in your case. Thank you for understanding.