espressif/esp-modbus #103: Packet Failure on Send Request – Incorrect Command Sent with Serial --> mb_master (ESP-IDF v5.4.0)

SMledpoint, 2025-03-10:
Description:
I’m experiencing an issue with the Modbus master operation when running on ESP-IDF v5.4.0. During a read coils operation, the master fails to receive a proper response. The log shows that a timeout occurs (ESP_ERR_TIMEOUT, 0x107) and, upon inspection with a logic analyzer, the transmitted packet appears malformed. The analyzer output indicates extra bytes (reported as “Message too long”) and an incorrect CRC.

Environment:

ESP-IDF Version: v5.4.0
Modbus Master Library: [Serial mb_master]
UART Configuration:
Port: 2
Baud Rate: 9600
Parity: NONE

> Log Output:

D (731) intr_alloc: Connected src 24 to int 3 (cpu 0)
D (735) app_start: Starting scheduler on CPU0
I (740) maD (740) intr_alloc: Connected src 25 to int 1 (cpu 1)
D (740) app_start: Starting scheduler on CPU1
in_task: Started on CPU0
D (750) heap_init: New heap initialised at 0x3ffe0440
D (750) heap_init: New heap initialised at 0x3ffe4350
D (760) intr_alloc: Connected src 16 to int 5 (cpu 0)
I (760) main_task: Calling app_main()
I (760) MASTER_LIB: Initializing Modbus master:
I (770) MASTER_LIB: UART PORT:2, BAUD:9600, PARITY:NONE
I (770) MASTER_LIB: PIN TX:23, RX:16
D (780) MB_PORT_COMMON: vMBPortSetMode: Port enter critical.
D (780) MB_PORT_COMMON: vMBPortSetMode: Port exit critical
D (790) MB_PORT_COMMON: eMBMasterRTUInit: Port enter critical.
I (790) uart: queue free spaces: 20
D (800) intr_alloc: Connected src 36 to int 8 (cpu 0)
D (800) MB_PORT_COMMON: xMBMasterPortSerialInit Init serial.
D (810) MB_PORT_COMMON: eMBMasterRTUInit: Port exit critical
D (810) MB_PORT_COMMON: eMBMasterRTUStart: Port enter critical.
D (820) MB_PORT_COMMON: eMBMasterRTUStart: PortD (820) DBG: RTU timer, init FSM state.
D (820) MBM_TIMER: Timer mode: (0) triggered
 exit critical
D (830) MB_PORT_COMMON: 0:EV_MASTER_READY
I (830) MASTER_LIB: Modbus master initialized successfully
D (1840) MASTER_LIB: Starting new Modbus operation cycle
I (1840) MASTER_LIB: Attempting READ - CID:0 Relay1 (Slave:1 Coil:0x0000)
D (1840) MB_PORT_COMMON: xMBMasterRunResTake:Take MB resource (500 ticks).
D (1840) MB_PORT_COMMON: 1230907:EV_MASTER_FRAME_TRANSMIT
D (1850) POLL transmit buffer: 01 00 00 00 01
D (1850) MB_PORT_COMMON: eMBMasterRTUSend: Port enter critical.
D (1860) MB_PORT_COMMON: eMBMasterRTUSend: Port exit critical
D (1860) MB_PORT_COMMON: xMBMasterPortSerialSendRequest default
D (1870) MB_PORT_COMMON: vMBMasterPortTimersRespondTimeoutEnable Respond enable timeout.
D (1880) MB_MASTER_SERIAL: MB_TX_buffer sent: (9) bytes.
D (1880) MB_PORT_COMMON: vMBMasterRxSemaRelease:RX semaphore is free.
D (1890) MB_PORT_COMMON: 1230907:EV_MASTER_FRAME_SENT
D (1890) POLL sent buffer: 01 00 00 00 01
D (2880) MBM_TIMER: Timer mode: (1) triggered
D (2880) MB_PORT_COMMON: 1230907:EV_MASTER_ERROR_PROCESS
D (2880) MB_PORT_COMMON: vMBMasterErrorCBRespondTimeout:Callback respond timeout.
D (2880) MB_PORT_COMMON: Transaction (1230907), processing time(us) = 1033744
D (2890) MB_PORT_COMMON: eMBMasterWaitRequestFinish: returned event = 0x100
D (2890) MB_CONTROLLER_MASTER: mbc_serial_master_get_parameter: Bad response to get cid(0) = ESP_ERR_TIMEOUT
E (2900) MB_CONTROLLER_MASTER: mbc_master_get_parameter(75): Master get parameter failure, error=(0x107) (ESP_ERR_TIMEOUT).
E (2920) MASTER_LIB: READ FAILED - CID:0, err=0x107 (ESP_ERR_TIMEOUT)

Expected Behavior:
The master should send a correctly formatted Modbus RTU request that adheres to the specification.
A proper response should be received from the slave, with valid CRC and data.

Actual Behavior:
The transmitted packet appears malformed (extra “Message too long” segments and incorrect CRC).
The master encounters a timeout (ESP_ERR_TIMEOUT) due to a bad or absent response.
Questions/Request for Assistance:

Is this behavior a known issue with the Modbus master library when used with ESP-IDF v5.4.0?
Could there be a configuration or timing issue (e.g., with the RTU timer or UART settings) that might cause the packet to be malformed?
Are there any recommended workarounds or updates available to address this problem?
I appreciate any guidance or insights you might provide to help resolve this issue. Please let me know if additional logs or configuration details are required.

alisitsyn, 2025-03-11:
@SMledpoint , thank you for update. On the pictures above I can see only the Modbus master request: `01 01 00 00 00 01 FD CA` which is correct and no any response from your slave. So, the master handles the request correctly with the timeout. Please check your device service manual to setup it accordingly. If you still think the master handles the response from slave incorrectly, please provide the actual data of master request and slave response on A, B or TXD, RXD pins accordingly.
Thank you.

SMledpoint, 2025-03-11:
@alisitsyn 

Thank you for responding. The packet issue was due to faulty slave hardware. After replacing the hardware, it is now able to send the correct packets without any errors. Thank you!

closed 2025-03-11 by SMledpoint

alisitsyn, 2025-03-12:
@SMledpoint , thank you for update. Good luck with your project!