espressif/esp-modbus #104: How to disable mDNS integration?

esp-dev, 2025-03-11:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hi,

I have a question regarding mDNS integration, how to disable it (in esp-modbus) if I already configure DNS in my part of the code?

alisitsyn, 2025-03-11:
Hi,

(the below is related to esp-modbus v2.x.x)
You do not need to disable the mDNS in the esp-modbus stack specifically. The stack is able to recognize if this optional dependency enabled for your project and will use the related functionality only if you added the mDNS dependency into your project. In this case if the mDNS dependency is not enable but you try to configure stack using the mdns host names in the configuration of your object the stack will report error. The dependency is resolved on CMAKE layer.

esp-dev, 2025-03-11:
Hi,

thx for reply.

I already use esp-mdns in the project for something else, I would like the modbus stack not to perform independent mdns configuration because I want to customize it accordingly. In addition, the modbus stack in the mdns context needs a pointer to the interface, which is not convenient and providing a NULL value causes a crash. The only solution I have found at the moment is to move from managed_components to components of the esp-modbus directory and comment out target_compile_definitions(${COMPONENT_LIB} PUBLIC -DMB_MDNS_IS_INCLUDED) which is not very elegant.

alisitsyn, 2025-03-12:
Hi,

> I already use esp-mdns in the project for something else, I would like the modbus stack not to perform independent mdns configuration because I want to customize it accordingly. 

The mdns functionality designed to be transparent for esp-modbus. So, you need to disable the mdns for esp-modbus but want to use in your project? I can consider some update for this but need to understand the issue in your project better.

> In addition, the modbus stack in the mdns context needs a pointer to the interface, which is not convenient and providing a NULL value causes a crash.

Could you describe the issue in more details with the logs? We can talk about possible solution then.

esp-dev, 2025-03-14:
Hi @alisitsyn,

so I wanted to disable everything related to mDNS in the library, because in our project espressif-mdns is already integrated and used for example to advertise the http service. So it is not a problem for us to add modbus advertising there too. Moreover, our project requires that the modbus service can be stopped and its port changed using configuration without restarting the system, so the service must be dynamically added and deleted from mDNS, therefore it is better to support mDNS externally (in our project). BTW (we use the modbus library in slave tcp mode) the tcp socket accepting connections is not closed by deinit/delete duncrionI don’t remember name, which prevents it from being restarted becouse port is in use.

alisitsyn, 2025-03-14:
Hi @esp-dev,

Thank you for update. This explains your project better and some investigation and refactoring is planned to provide better way for mDNS integration in the esp-modbus. I am going to do this once possible.  For now my proposal is the simple change in the esp-modbus library as a kconfig option which would allow to explicitly disable/enable esp-modbus - mDNS integration. Will this solve the issue in your project?

closed 2025-09-30 by alisitsyn