espressif/esp-modbus #105: FMB_TCP_CONNECTION_TOUT_SEC causes an overflow

femtotech, 2025-03-26:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

The SDK asks to input the number of seconds for the Modbus connection timeout (`CONFIG_FMB_TCP_CONNECTION_TOUT_SEC`). I didn't find the allowable range, but the internal #define (`MB_TCP_DISCONNECT_TIMEOUT`) is expressed in us and used by an `int64` variable [here](https://github.com/espressif/esp-modbus/blob/master/freemodbus/tcp_slave/port/port_tcp_slave.c#L644).

### Actual behavior (suspected bug)

The `MB_TCP_DISCONNECT_TIMEOUT` #define ([here](https://github.com/espressif/esp-modbus/blob/master/freemodbus/tcp_slave/port/port_tcp_slave.c#L63)) states:

    #define MB_TCP_DISCONNECT_TIMEOUT       ( CONFIG_FMB_TCP_CONNECTION_TOUT_SEC * 1000000 ) // disconnect timeout in uS

**This would cause an overflow** if, for example, one sets `CONFIG_FMB_TCP_CONNECTION_TOUT_SEC = 3600`.

### Error logs or terminal output

```shell
This is the output when `CONFIG_FMB_TCP_CONNECTION_TOUT_SEC = 3600`

I (58248) MB_TCP_SLAVE_PORT: Socket (#55), accept client connection from address: 192.168.2.62
E (58250) MB_TCP_SLAVE_PORT: Client 0, Socket(#55) do not answer for 14 (us). Drop connection...
I (61228) MB_TCP_SLAVE_PORT: Socket (#55), accept client connection from address: 192.168.2.62
E (61232) MB_TCP_SLAVE_PORT: Client 0, Socket(#55) do not answer for 13 (us). Drop connection...
I (64299) MB_TCP_SLAVE_PORT: Socket (#55), accept client connection from address: 192.168.2.62
E (64301) MB_TCP_SLAVE_PORT: Client 0, Socket(#55) do not answer for 14 (us). Drop connection...
```

### Steps to reproduce the behavior

1. set `CONFIG_FMB_TCP_CONNECTION_TOUT_SEC = 3600`
2. run a Modbus Client 
3. try to connect

### Project release version

1.0.17

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 24.10

### Shell

Bash

### Additional context

I would propose the following patch:

    #define MB_TCP_DISCONNECT_TIMEOUT       ( CONFIG_FMB_TCP_CONNECTION_TOUT_SEC * 1000000UL )

In this way, you can set `CONFIG_FMB_TCP_CONNECTION_TOUT_SEC = 3600` and it will work fine.

alisitsyn, 2025-03-27:
Hello, @femtotech,

Thank you for this report. The maximum time has never been checked and it was expected to have something like 5 seconds at maximum waiting for connections. The maximum value for (int64_t) = 9223372036854775807. I confirm the issue in the  [line]( https://github.com/espressif/esp-modbus/blob/master/freemodbus/tcp_slave/port/port_tcp_slave.c#L644) where the constant is treated as an integer instead of (int64_t) and caused overflow as stated by the compilation warning:
`esp-modbus/freemodbus/tcp_slave/port/port_tcp_slave.c:63:78: warning: integer overflow in expression of type 'int' results in '-694967296' [-Woverflow]`
Your change solves the issue with incorrect comparison. However, I can not change this releasing the new component because the v1.0.17 is out of support because of [v2 is in the focus](https://github.com/espressif/esp-modbus/discussions/99). In spite of this I will try to fix this when possible in the master with low priority.

alisitsyn, 2025-04-25:
Merged to master with commit 8c62835036616612fad2b6a36501b9a1af2582d5

closed 2025-04-25 by alisitsyn