espressif/esp-modbus #106: modbus_params.h

tangqingmiao, 2025-03-27:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

master.c.obj -MF esp-idf/main/CMakeFiles/__idf_main.dir/serial_master.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/serial_master.c.obj   -c ../main/serial_master.c
../main/serial_master.c:9:10: fatal error: modbus_params.h: No such file or directory
    9 | #include "modbus_params.h"  // for modbus parameters structures
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.

Godoy-Octavio, 2025-03-27:
Hello, @tangqingmiao 

Thank you for reporting this issue. The file modbus_params.h is included in ../examples/mb_example_common

In order of including this file to your project modify the ../mb_serial_master/main/idf_component.yml file to include the mb_example_common dependency using the path:

    ${IDF_PATH}/examples/protocols/modbus/mb_example_common 

Once modified, building the app from scratch should work without problems:
 
    idf.py fullclean && idf.py build 

For more details on Modbus configuration and parameter mapping, please refer to:

[ESP-Modbus Serial Master Example](https://github.com/espressif/esp-modbus/blob/main/examples/serial/mb_serial_master/README.md)

[ESP-IDF Modbus Documentation - Data mapping](https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/api-reference/protocols/modbus.html#messaging-model-and-data-mapping)

Let us know if you encounter any further issues!

alisitsyn, 2025-04-23:
The issue is closed. Feel free to reopen.

closed 2025-04-23 by alisitsyn